butterwecksolutions/Spotmarket-Switcher #1: Solarprognose via https://www.solarprognose.de

bacanol, 2023-01-09:
Hallo @christian1980nrw,

finde Dein Skript sehr gut! Funktioniert auch gut bei mir, meinst Du man könnte die API von https://www.solarprognose.de dort einbauen? Die ist etwas IMHO etwas granularer einstellbar auf die eigene PV-Anlage?

Viele Grüße

christian1980nrw, 2023-01-09:
Hallo Nico @bacanol , ich schaue mir die Solarprognose API mal an. Die kannte ich bisher nicht. Ist die auch weltweit nutzbar?
Die bisherige hat den Vorteil, dass man nicht viel machen musste und die passenden Tageserträge direkt von der API kommen.

Gruß!
Christian

christian1980nrw, 2023-01-09:
Hallo Nico @bacanol , 
also einen großen Unterschied machen die APIs nicht. Das lohnt die Mühe nicht. Erstmal ist die Solarprognose API nur für Deutschland gedacht, mein Script soll jedoch in der ganzen EU genutzt werden können. 
Außerdem ist es viel zu kompliziert umgesetzt. Das macht keinen Spaß das zu konfigurieren. 
Ich habe auch mal nachgerechnet. Heute habe ich laut Visualcrossing in meiner Stadt 2.8 Megajoule pro Quadratmeter Ertrag. Also wenn ich das mit meiner Modulfläche multipliziere, komme ich bei 15% Effizienz (laut Datenblatt der Zellen etwa 19%, ich nehme jedoch 15% aufgrund von Abzügen für die Rahmen) nach Umrechnung von Megajoule auf kWh in etwa auf denselben Ertrag wie von Solarprognose berechnet.
Das einzige ist die nicht berücksichtige Verschattung. Beide APIs stimmen deshalb in nahezu identischer Höhe nicht mit der Realität überein. Es wäre vielleicht eine Idee im Script für jeden einzelnen Monat einen anderen Abbruchwert eingeben zu können, dann kann man das in etwa mit berücksichtigen.

Gruß!
Christian

closed 2023-01-09 by christian1980nrw

bacanol, 2023-01-10:
Moin @christian1980nrw,

Danke für Dein Feedback. Ich hatte es für sehr interessant gehalten, da es nicht pauschal von einer Südausrichtung der Anlage ausgeht und dabei dann auch eine 10 Tagesprognose bekommt. Ich denke auch eine Relation zum Eigenverbrauch interessant sein könnte, um diesen dann mit einzubeziehen. 

Ich werde, bei entsprechender Zeit einmal schauen, wie man das einbauen könnte und Ann ggf einen PR zur Verfügung stellen.

Viele Grüße
Nico

christian1980nrw, 2023-01-10:
@bacanol So einfach ist das nicht, da man idealerweise auch den Winkel berücksichtigen muss. Da je nach Land und Standort andere Parameter gelten, wäre es für dieses Script etwas übertrieben. Aber monatsweise andere Verlustparameter bis zum Abbruch einzugeben werde ich vielleicht noch umsetzen. 
siehe auch https://web.stanford.edu/group/efmh/jacobson/Articles/I/TiltAngles.pdf