butterwecksolutions/Spotmarket-Switcher #3: Starts charging randomly - controller.sh: line 223: printf: e15: invalid number

bacanol, 2023-01-23:
Moin,

ich glaube es liegt ein Fehler im Skript vor. Bei bestimmten Preiskonstellationen kommt als "Current price" eine NULL heraus und er startet das Laden ohne Rücksicht: 
`'Current price is' ' Cent/kWh netto.'`
Parallel gibt es eine Fehlermeldung im Log:
`line 223: printf: e15: invalid number`

Viele Grüße
Nico

christian1980nrw, 2023-01-23:
Moin Nico, nutzt Du die neueste Version? 
Kannst Du die Dateien aus dem Temp-Verzeichnis zum Zeitpunkt des Fehlers und Deine Konfiguration hier anhängen? 
Normalerweise darf keine Fehlermeldung im Log auftauchen.

Gruß!
Christian

bacanol, 2023-01-23:
Moin Christian,

ich nutze die 1.7 aktuell, werde aber gleich mal aktualisieren. 

Der ursächliche Fehler ist allerdings wohl dieser hier:
`controller.sh: line 201: 2*08: value too great for base (error token is "08")`

Und die Zeile 201 im Code ist:
`function get_awattar_prices { current_price=$(sed -n $((2*$(TZ=$TZ date +%H)+39)){p} $file1 | grep -Eo '[+-]?[0-9]+([.][0-9]+)?' | tail -n1);`

Viele Grüße
Nico

bacanol, 2023-01-23:
Mit 1.8 ist der Fehler noch enthalten:
`controller.sh
Maybe we are still charging from last script runtime. Stopping scheduled charging.
0
Awattar today-data is up to date.
Awattar tomorrow-data is up to date.
./tibber-controller.sh: line 219: 2*09: value too great for base (error token is "09")
./tibber-controller.sh: line 248: printf: e15: invalid number
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100    78  100    78    0     0    114      0 --:--:-- --:--:-- --:--:--   114
/tmp/expected_solarenergy.csv downloaded
./tibber-controller.sh: line 268: printf: e15: invalid number
./tibber-controller.sh: line 269: printf: e15: invalid number
date: invalid date '1970-01-01 '
date: invalid date '1970-01-01 '
./tibber-controller.sh: line 274: (( - )) / 60: syntax error: operand expected (error token is ")) / 60")
Please verify correct system time and timezone:
Mon Jan 23 09:02:05 CET 2023
Current price is  Cent/kWh netto.
Lowest price will be 14.75900 Cent/kWh netto.
The average price will be 20.2734 Cent/kWh netto.
Highest price will be 27.02200 Cent/kWh netto.
Second lowest price will be 14.84000 Cent/kWh netto.
Third lowest price will be 14.89600 Cent/kWh netto.
Fourth lowest price will be 14.90000 Cent/kWh netto.
Fifth lowest price will be 14.98800 Cent/kWh netto.
Sixth lowest price will be 15.03200 Cent/kWh netto.
Sunrise today will be  and sunset will be . Suntime will be  minutes.
Solarenergy today will be  megajoule per sqaremeter with  percent clouds.
Solarenergy tomorrow will be  megajoule per squaremeter with  percent clouds.
Difference between highest price and current price is greater than 23.3%.
Charging makes sense. Executing 1 hour charging.
0`

bacanol, 2023-01-23:
Vermutung ist, dass hier mit Octal-Zahlen gearbeitet wird und 09 sowie 08 ja zu hoch sind. Ich denke bei der Bash müsste man mitgeben, dass man Dezimalzahlen nutzen möchte.

bacanol, 2023-01-23:
`sed -n $((2*$(TZ=$TZ date +%H)+39)){p} /tmp/awattar_today_prices.yaml
-sh: 2*08: value too great for base (error token is "08")`

christian1980nrw, 2023-01-23:
Das Problem ist das mit 08 anstatt 8 gearbeitet wird. Eine Bash kann eine 08 nicht verarbeiten. Der Aufruf des date Befehlt muss anstelle mit +%H mit date +%k erfolgen. Ich korrigiere es wenn es in der aktuellen Version noch falsch ist.

bacanol, 2023-01-23:
> Das Problem ist das mit 08 anstatt 8 gearbeitet wird. Eine Bash kann eine 08 nicht verarbeiten. Der Aufruf des date Befehls muss anstelle mit +%H mit date +%k erfolgen. Ich korrigiere es heute Abend wenn es in der aktuellen Version noch falsch ist.

%k funktioniert hier. 👍 Danke!

closed 2023-01-23 by bacanol