butterwecksolutions/Spotmarket-Switcher #4: does it respect Awattar time to configure the scheduler?

nudelkiste, 2023-02-12:
Hi,

just trying your script and it seems to work fine as it switches on/off the first scheduler.
All I wonder is: does it not configure the dates and time of the scheduler according to what it read by Awattar?

Here my output to explain my question a bit more in detail:
`root@einstein:~/scripts# ./controller.sh 
Maybe we are still charging from last script runtime. Stopping scheduled charging. Battery SOC is at 40 %.
0
Awattar today-data is up to date.
Awattar tomorrow-data is up to date.
Please verify correct system time and timezone:
Sun Feb 12 22:16:17 CET 2023
Current price is 15.19100 Cent/kWh netto.
Lowest price will be 10.80200 Cent/kWh netto.
The average price will be 14.9902 Cent/kWh netto.
Highest price will be 22.18000 Cent/kWh netto.
Second lowest price will be 11.61700 Cent/kWh netto.
Third lowest price will be 11.84600 Cent/kWh netto.
Fourth lowest price will be 11.88900 Cent/kWh netto.
Fifth lowest price will be 12.00000 Cent/kWh netto.
Sixth lowest price will be 12.00000 Cent/kWh netto.
Difference between highest price and current price is greater than 23.3%.
Charging makes sense. Executing 1 hour charging. Battery SOC is at 40 %.
0
root@einstein:~/scripts# 
`

This sounds like "starting one hour now", but my scheduler is configured as follows:
[image]
So it will load for 3 hours and not now but within a about 1h40m 

I don´t see how the script helps to load during low price when the schedule does not change the start-time nor the duration?

Let me extend the question about how to change start-time and duration in therms of whether conditions.
Today my remaining capacity won't be suitable to reach sunrise. So it would be great to load in the remaining time when the price would be lowest within that timeframe. To be more precise, not even sunrise is interesting, but the time to when the sunshine is expected due to weather conditions!

christian1980nrw, 2023-02-12:
Mahlzeit, so funktioniert das nicht. Du kannst im Zeitplan am besten eine Dauer von 23 Stunden 59 Minuten konfigurieren und ihn nach Erstellung deaktivieren. Das Script schaltet ihn dann ein wenn z.B. der günstigste Preis ist. Je nachdem was Du im Script konfiguriert hast. Sobald der Preis nicht mehr am günstigsten ist, wird der Ladeplan ausgeschaltet vom Script. Der Ladeplan im Victron dient also nur als Ein- und Ausschalter und kann daher über den ganzen Tag reichen. Gruß! Christian

closed 2023-02-12 by christian1980nrw

christian1980nrw, 2023-02-12:
Du könntest im Victron (siehe Dein Screenshot) "Stopp bei Ladezustand" aktivieren und dort z.B. 50% oder weniger eintragen. Dann würdest Du den Rest für Solarstrom reservieren. Außerdem kannst Du die Solarwetter-API im Script aktivieren und abhängig vom erwarteten Solarertrag gar nicht aus dem Netz laden.

nudelkiste, 2023-02-13:
Irgendetwas an der config vom Script muss ich übersehen oder falsch verstanden haben. Es funktioniert ganz anders als erwartet.
**Beschreibung des Ziels:**
Irgendwann in denn kommenden zwei Stunden, also knapp vor Sonnenaufgang um , werde ich vermutlich auf 20% Akku gefallen sein und damit keine Reserven mehr haben.
[image]
Ausserdem kommt die Sonne bei mir erst 1,5h nach Sonnenaufgang über den Berg.
Also wollte ich zu günstigen Zeiten nachladen, laut Awattar zwischen 3:00 und 4:00 Uhr
[image]
[image]

**Abweichung vom Ziel:**
Das script wollte aber direkt um 22:00 Uhr schon das Laden starten, vermutlich weil zu dem Zeitpunkt folgende Bedingung erfüllt war: "Difference between highest price and current price is greater than 50%."

**Problem dabei**
Der "highest" price kann voll in den Sonnenstunden liegen und interessiert deshalb gar nicht.
Dadurch, dass der aktuelle Prozentsatz nicht berücksichtigt wird, genausowenig wie die "Distanz" bis zum Sonnenaufgang und auch nicht die notwendige zuzuladende Strommenge, dadurch schaltet das Script nicht zu den optimalen Zeitpunkten, sondern "nur" zu den Zeitpunkten wo es 

**Gedankenexperiment zur Verdeutlichung:**
Ein relativ stabiler, Preis über 23h mit einem heftigen Preispeak während einer Stunde
-> Das würde dazu führen, dass während 23h das script den geplanten Ladevorgang einschalten würde.

Würde ich Deinem Vorschlag folgen ("im Zeitplan am besten eine Dauer von 23 Stunden 59 Minuten konfigurieren") und eine Maximalrate von z.B. 40% angegeben haben, so würde zwar das Laden stoppen, nicht aber der Ladeplan. Das führt dazu, dass weiterhin der Verbrauch aus dem Netz gezogen wird um den SoC zu halten. Das sieht dann so aus:
[image]
Etwas unschön, da ich ja nur um 3:00 Uhr nur 10 Min nachladen wollte. Das reicht mir um mit 8kW Ladeleistung von 30% wieder auf 40% SoC zu steigen und so bis zum Sonnenaufgang durchzuhalten.

christian1980nrw, 2023-02-13:
Difference between highest price and current price is greater than 50%.
Die 50% sind viel zu hoch,  Du solltest besser den Standard Wert belassen. Es geht dabei nur darum nicht zu laden wenn der höchste Preis zu günstig ist als der zum Ladezeitpunkt einschließlich Ladeverluste. Denn dann braucht man nicht zu laden und kann auch zu Tageskursen verbrauchen.

Der SOC oder Verbrauch interessiert das Script aktuell herzlich wenig, es werden nur die Preise und ggf. das Solarwetter beachtet. Der SOC wird nur zur Information geloggt,  ggf. mache ich in zukünftigen Versionen was damit. Du solltest also soviel laden das du ungefähr für den Tag über die Runden kommst. Du solltest mal versuchen nur den niedrigsten Preis mit 1 zu aktivieren und alles andere auf 0 zu lassen.