butterwecksolutions/Spotmarket-Switcher #5: integration with cerbo gx

Apollo0584, 2023-03-12:
hello,

did you check https://github.com/tfranssen/dynamic-ess ?

currently testing your script on cerbo Gx with latest VenusOS installed.
API connection for Belgium. awaiting token from provider.
possible to activate via schedule for end-user?

christian1980nrw, 2023-03-12:
No, i didnt know this projekt. I will check it later, thanks. Can you explain your question? I dont get it. The script is running automatically every full hour after installation, there is no GUI for a End User. For Belgium you need the Entsoe API, please read the notes inside the controller.sh how to get the api key.

closed 2023-03-12 by christian1980nrw

Apollo0584, 2023-03-12:
The api from entsoe is not for Belgium. Request needed via the local entity Elia with a token.

My question was if it could be possible with your script to trad and set some parameters,
using the existing menu in the OS (assigning schedule nr4 for example) and take those values to adjust the script. 
That way the user can set multiple fixed values when scripting runs in the background.

christian1980nrw, 2023-03-12:
Are you sure? Prices from belgium are shown there.
[image]
If you need other data sources you have to tell me how to get free access to them so that i can have a look.