butterwecksolutions/Spotmarket-Switcher #7: Date Solarenergy issue

misterclownfish, 2023-07-29:
Hy Christian, 

Great project!

I just have die following issues with the date from the solar energy csv:
```
Maybe we are still charging from last script runtime. Stopping scheduled charging.
Battery SOC is at 29 %.
0
Awattar today-data is up to date.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100    51  100    51    0     0     80      0 --:--:-- --:--:-- --:--:--    80
/tmp/expected_solarenergy.csv downloaded
/data/etc/Spotmarket-Switcher/controller.sh: line 282: printf: e15: invalid number
/data/etc/Spotmarket-Switcher/controller.sh: line 283: printf: e15: invalid number
date: invalid date '1970-01-01 '
date: invalid date '1970-01-01 '
/data/etc/Spotmarket-Switcher/controller.sh: line 288: (( - )) / 60: syntax error:
operand expected (error token is ")) / 60")
Please verify correct system time and timezone:
Sat Jul 29 11:39:42 CEST 2023
Current price is 6.37500 Cent/kWh netto.
Lowest price will be 1.82000 Cent/kWh netto.
The average price will be 7.66754 Cent/kWh netto.
Highest price will be 10.30900 Cent/kWh netto.
Second lowest price will be 1.96700 Cent/kWh netto.
Third lowest price will be 4.04800 Cent/kWh netto.
Fourth lowest price will be 4.91800 Cent/kWh netto.
Fifth lowest price will be 5.86600 Cent/kWh netto.
Sixth lowest price will be 5.86600 Cent/kWh netto.
Sunrise today will be  and sunset will be . Suntime will be  minutes.
Solarenergy today will be  megajoule per sqaremeter with  percent clouds.
Solarenergy tomorrow will be  megajoule per squaremeter with  percent clouds.

```

misterclownfish, 2023-07-29:
Ha, I overread the account for Visual crossing...
Now it works like a charm :-)

misterclownfish, 2023-07-29:
Works!

closed 2023-07-29 by misterclownfish

christian1980nrw, 2023-07-29:
:-)

misterclownfish, 2023-11-07:
> Same Problem: date: invalid date '1970-01-01 ' date: invalid date '1970-01-01 ' controller.sh: line 624: ( - ) But dont understand the solution: "I overread the account for Visual crossing..."
> 
> thank you for your help!

Hy, you need an account for the forecast API “Visual crossing“.
Go to the homepage and register, then place account info (“token“) into the script.

christian1980nrw, 2023-11-07:
@bauer1980 Did you open a free visualcrossing account and put your API key into the config?

christian1980nrw, 2023-11-07:
The process is described in the config.txt comments.

bauer1980, 2023-11-07:
> @bauer1980 Did you open a free visualcrossing account and put your API key into the config?

thank you, i understand!!