butterwecksolutions/Spotmarket-Switcher #66: Consecutive "good" hours should not be interrupted.

smoe, 2023-09-13:
From what I have understood, in its current implementation all activities (charging, switches) are switched off whenever the control script starts at every hour. Also, there is a long sleep that turns things off after these were switched on.

I would like to change that behaviour to switch off only when the following hour is not good.

christian1980nrw, 2023-09-13:
Consecutive “good” hours should not be interrupted is a good idea. In this context, you will probably think about running everything without the crontab. My suggestion would then be to switch to a minute based calculation instead of an hourly basis, as there are also dynamic tariffs based on 15 minute intervals. In addition, it will then / later be easier to calculate the minimum running time of consumers based on minutes.

smoe, 2023-09-14:
My immediate thought was to just not shut down what would be switched on 15 seconds later again - the system should just quickly check what it would decide for "date+1h" and if that decision says "good", then keep it on. 

There is some extra thoughts to this - some processes you just do not want to interrupt - washing your clothes being one of them. So, you may have initiated the start when the energy is low but then keep it running until the process has finished, disregarding the costs for power since otherwise you may waste some water and energy to get the process running again.

Some smart switches are also metering the energy passed through, so for a switch to switch off for some applications it may be desirable to check that for the last 5 minutes no notable consumption was observed.

christian1980nrw, 2023-09-14:
When I designed the script, the initial focus was on charging the battery. What do you think about first removing the crontab based start behavior and moving it to a configurable 60 or 15 minute-based main loop within Controller.sh. After that its maybe easier to build in a more intelligent logic to start and stop consumers.

smoe, 2023-09-15:
Yes, can do, not ultimately convinced that this would be required for the change I suggested.

christian1980nrw, 2023-09-29:
Fix is untested. Should work better now and without break we did before.

closed 2023-09-30 by christian1980nrw