butterwecksolutions/Spotmarket-Switcher #81: seit Venus OS 3.1 keine Ladeplanung über Spotmarketswitcher möglich?!

alex90alex, 2023-10-12:
Hallo zusammen,

ich habe vor einigen Tagen die neue Firmware Venus OS 3.1 bzw. 3.11 aufgespielt. Das dynamische ESS verhält sich bei mir leider nicht so intelligent wie ich es gerne hätte. Mit aktivem dynamischen ESS wird unnötig der Akku entladen oder teilweise der Akku in der Nacht voll gehalten, trotz hohen Stromkosten (Tibber). Die Logik vom dynamischen ESS scheint aktuell noch nicht für mich zu passen. Daher würde ich gerne wieder die Funktionen vom Spotmarketswitcher verwenden. 
Wenn ich die Einstellung von dynamic ESS wieder auf "off" stelle, erfolgt leider die Ladeplanung über den Spotmarketswitcher nicht mehr. Hat jemand hierfür eine Problemlösung? In den letzten Monaten hat der Spotmarketswitcher super funktioniert. :-) 

Danke für eure Hilfe.

Viele Grüße
Alex

christian1980nrw, 2023-10-12:
Hi, als erstes würde ich Dir empfehlen, die neueste Spotmarket-Switcher Version zu installieren, da seit neuestem auch die Batteriekosten berücksichtigt werden. Dann kannst Du in /tmp/spotmarket-switcher.log kontrollieren ob versucht wurde zu laden.
Außrdem könntest Du auf der Shell mal testen ob sich Dein erster ESS Ladeplan aktivieren lässt mit:

 "dbus -y com.victronenergy.settings /Settings/CGwacs/BatteryLife/Schedule/Charge/0/Day SetValue -- 7"

sowie wieder deaktivieren mit
"dbus -y com.victronenergy.settings /Settings/CGwacs/BatteryLife/Schedule/Charge/0/Day SetValue -- -7"

Gruß!
Christian

alex90alex, 2023-10-14:
Hi Christian,
danke dir. Die neue Version von Spotmarket-Switcher habe ich installiert. In der Log Datei werden wohl die richtigen Befehle gespeichert: (Ich habe als Grenze 19 Cent/kWh)
Sat Oct 14 18:00:02 CEST 2023
Current price is 0.3049 EUR/kWh total price.
I: Current price is too high. Abort.

Leider wird die Ladeplanung bei teuren Preisen nicht mehr abgebrochen. 
Über den Befehl: "dbus -y com.victronenergy.settings /Settings/CGwacs/BatteryLife/Schedule/Charge/0/Day SetValue -- -7" lässt sich der Ladeplan deaktivieren. 
Wenn ich jetzt von Hand den Ladeplan aktiviere müsste das Skript doch spätestens zur nächsten vollen Stunde den Plan wieder deaktivieren, somit heute Abend zu jeder vollen Stunde.
Ich werde noch etwas beobachten und melde mich dann wieder. 

Viele Grüße
Alex

christian1980nrw, 2023-10-14:
Wenn zwei aufeinander folgende Stunden die Bedingungen erfüllen, wird nicht mehr zwischendurch kurz abgeschaltet wie es in alten Versionen war.

Gruß Christian

alex90alex, 2023-10-15:
Komischerweise scheint sich der Zeitplan beim Unterschreiten der Preisgrenze automatisch zu aktivieren, aber beim Überschreiten wird der Zeitplan nicht beendet. Ich habe die Vermutung, dass ich das Update nicht richtig installiert habe. Wie deinstalliert man den Spotmarketswitcher richtig? Ich habe nur die beiden Befehle aus deiner Beschreibung ausgeführt und als ein Fehler beim Starten erschien (Datei existiert bereits) habe ich die Datei auf die Spotmarketswitcher von Hand gelöscht und den Befehl: "sh victron-venus-os-install.sh" erneut ausgeführt. Sorry meine Linux Kenntnisse sind überschaubar. ;-).

christian1980nrw, 2023-10-15:
Ganz einfach:

cd /path/to/save/
mv ./controller.sh ./controller-alt.sh
wget https://raw.githubusercontent.com/christian1980nrw/Spotmarket-Switcher/main/scripts/controller.sh 
chmod +x ./controller.sh 

vi /path/to/save/controller.sh und die eigenen Parameter wieder eintragen.

ckvsoft, 2023-10-16:
Hast Du schon in der Logdatei geschaut? Da sollte ja stündlich was drinnen stehen was er vor hatte.

Lg Chris

christian1980nrw, 2023-10-16:
@alex90alex Es war ein Fehler im Script der Dein Problem vermutlich verursacht hat. 
Du kannst nochmal neu herunterladen, beachte aber dabei, dass die config nun in einer ausgelagerten Datei erwartet wird.
Danke an @ckvsoft fürs Gegenlesen des aktuellen Codes.

christian1980nrw, 2023-10-17:
Issue should be fixed at the master and dev branch. Still testing.

alex90alex, 2023-10-17:
Okay. Ich habe nun die unten stehenden Einstellungen übernommen um das Umschalten der Logik zu testen. 
Komischerweise wird der Ladeplan nicht aktiviert, obwohl der aktuelle Preis weit unter der Grenze von 490 Ct/kWh liegt.
Folgende Meldung erscheint wenn ich controller.sh ausführe. 

Current price is 0.3602 EUR/kWh total price.
Lowest price will be 0.2816 EUR/kWh total price.
The average price will be 0.319396 EUR/kWh total price.
Highest price will be 0.3919 EUR/kWh total price.
Second lowest price will be 0.2836 EUR/kWh total price.
Third lowest price will be 0.2888 EUR/kWh total price.
Fourth lowest price will be 0.2901 EUR/kWh total price.
Fifth lowest price will be 0.293 EUR/kWh total price.
Sixth lowest price will be 0.2935 EUR/kWh total price.
I: Current price is too high. Abort.

Einstellungen:
 # Solar Charger Setup (tested with Victron Venus OS)
use_victron_charger=1                         # please activate with 1 or deactivate this charger-type with 0
energy_loss_percent=0.0                      # Enter how much percent of the energy is lost by the charging and discharging process.
battery_lifecycle_costs_cent_per_kwh=0.0 # If you are using a chinese 5KWh LifePo4 battery (4.5 KWh useable) with 6000 cycles and the rebuying price is at 1200 EUR, the calculation will be 1200*100/6000/4,5.
economic_check=2                              # Set to 0 disables the check.
                                              # Set to 1 or 2. Current price + energy loss + battery lifecycle costs will be compared with (1 = highest_price / 2 = average_price) and aborted if charging makes no sense.

  #Please change prices (always use Cent/kWh, no matter if youre using Awattar (displaying Cent/kWh) or Entsoe API (displaying EUR/MWh) / net prices excl. tax).
stop_price=490.0    # stop above this price
start_price=490.0   # start below this price
feedin_price=18.5 # your feed-in-tariff of your solar system
energy_fee=15.3 # proofs of origin, allocations, duties and taxes (in case if stock price is at 0 Cent/kWh)
abort_price=490.0  # abort and never charge or switch if actual price is same or higher than this (Energy fees not included)

Zum Verständnis: 
Da meine EEG-Vergütung bei 18,5 Cent liegt, hatte ich die Grenze für den Spotmarketswitcher immer auf 19 Cent stehen. Somit konnte ich ein Entladen des Akkus bei niedrigen Preisen unterbinden und ggf. sogar günstiger als der entgangene Gewinn durch Solarstromeinspeisung den Hausspeicher laden. Beim Umschalten der Laderegelung meiner Wallbox, sobald der Strompreis von Tibber <19Cent/kWh würde normalerweise mein Hausspeicher leerlaufen. Vor meiner Ausprobiererei mit der Option "Dynamik ESS" bzw. der neuen Venus OS Version hat das mit dem Spotmarketswitcher gut funktioniert. Ist die Logik auch aktuell noch umsetzbar? :-)  Das wäre super.

ckvsoft, 2023-10-17:
Die Version von Gestern hat bei aWATTar noch einen Fehler, der wurde erst Heute behoben.

ckvsoft, 2023-10-17:
> Okay. Ich habe nun die unten stehenden Einstellungen übernommen um das Umschalten der Logik zu testen. Komischerweise wird der Ladeplan nicht aktiviert, obwohl der aktuelle Preis weit unter der [...]

Wie schauen die weiteren Einstellunge aus?

christian1980nrw, 2023-10-17:
Bei 
stop_price=490.0 # stop above this price
start_price=490.0 # start below this price
kann niemals gestartet werden. 
Das sind 4,90 EUR. Die Preise sind grundsätzlich in Cent einzugeben.
Gruß!
Christian

alex90alex, 2023-10-17:
[config.txt](https://github.com/christian1980nrw/Spotmarket-Switcher/files/12953467/config.txt)

Das ist meine config und log Datei. Die Spotmarketversion habe ich erst heute Abend heruntergeladen und gestartet.
[spotmarket-switcher.log](https://github.com/christian1980nrw/Spotmarket-Switcher/files/12953688/spotmarket-switcher.log)

Vielleicht lässt sich daraus etwas ableiten. Danke euch beiden.

christian1980nrw, 2023-10-17:
Probier mal folgendes: https://github.com/christian1980nrw/Spotmarket-Switcher/commit/7cf49618a2e2c8a96ffa027253ba99ea9bf28bcc

Wahrscheinlich ist es damit behoben.

alex90alex, 2023-10-17:
> Bei stop_price=490.0 # stop above this price start_price=490.0 # start below this price kann niemals gestartet werden. Das sind 4,90 EUR. Die Preise sind grundsätzlich in Cent einzugeben, egal was die API anzeigt. Gruß! Christian

Ich wollte damit ja erzwingen, dass er auf die Ladeplanung umschaltet. War ja nur als Test gedacht. Normalerweise würde ich hier 19 Cent eintragen.

christian1980nrw, 2023-10-17:
Du kannst Du die controller.sh nochmal neu ziehen, ich habe den kleinen Fix jetzt ins Main übernommen.
Bei mir wurde der Ladeplan aktiviert.

christian1980nrw, 2023-10-17:
Die Ausgabe ist aber noch Unsinn, aber daran nicht stören.
Victron scheduled charging is ON. Battery SOC is at 71 %. **Charging based on average price comparison makes sense.**
Turning OFF Fritz sockets.

alex90alex, 2023-10-17:
Was mir nun aufgefallen ist, dass die Tibber Preise nun Netto sind. Aber nun aktiviert sich der Ladeplan, aber er deaktiviert sich nicht mehr, wenn ich z.b. die Preisgrenze auf 5 Cent anpasse und das Skript starte müsste der Ladeplan doch aus gehen oder?

christian1980nrw, 2023-10-17:
Das kannst Du konfigurieren in der config.txt, ziemlich am Ende.
tibber_prices=total  # Set to
  #  "energy" to use the spotmarket-prices (default), set to
  #  "total" to use the total prices including taxes and fees, set to
  #  "tax" to use only the taxes and fees

ckvsoft, 2023-10-18:
> Probier mal folgendes: [7cf4961](https://github.com/christian1980nrw/Spotmarket-Switcher/commit/7cf49618a2e2c8a96ffa027253ba99ea9bf28bcc)
> 
> Wahrscheinlich ist es damit behoben.

Guten Morgen!

Da bin ich mir nicht sicher.

Der Hauptunterschied zwischen den beiden liegt in der Art und Weise, wie die logische UND-Operation repräsentiert wird:

    &&: Dies ist der übliche Weg, um eine logische UND-Operation in vielen Programmiersprachen, einschließlich Bash, darzustellen. Es wird verwendet, um zu überprüfen, ob beide Bedingungen wahr sind.

    -a: Dies ist auch eine logische UND-Operation, aber sie wird eher in traditionellen Testkonstrukten in Shell-Skripten verwendet. In Bash ist && innerhalb von doppelten eckigen Klammern [[ ... ]] bevorzugt, während -a innerhalb von einfachen eckigen Klammern [ ... ] oder ohne eckige Klammern in Verbindung mit test genutzt wird.

Im speziellen Kontext, innerhalb der doppelten Klammern (( ... )), ist && der bevorzugte und korrekte Weg, da diese Klammern arithmetische Ausdrücke evaluieren und && hier funktioniert wie in den meisten anderen Sprachen. Der Operator -a wäre in diesem Kontext nicht korrekt und könnte zu unerwarteten Ergebnissen führen.

Zusammengefasst sollte  bei der Verwendung von doppelten Klammern in Bash die erste Zeile (&&) verwenden.

christian1980nrw, 2023-10-18:
Die Version mit && lief nicht unter Venus OS. Oder man muss es anders bauen. @ckvsoft  

Gruß Christian

ckvsoft, 2023-10-18:
> Die Version mit && lief nicht unter Venus OS. Oder man muss es anders bauen. @ckvsoft
> 
> Gruß Christian

Kann ja nicht sein das es solche unterschiede gibt. Ich bau mir grad ein VenusOS Testsystem auf einen RPI4 auf und check das dann.

ckvsoft, 2023-10-18:
Testest Du noch? Oder Funktioniert es jetzt bei Dir?

Ich hab es jetzt auf VenusOS 3.11 im Testlauf, auch mit Tibber und es scheint zu funktionieren,

Du kannst aber gerne meine Dev Version  [Testen
](https://raw.githubusercontent.com/ckvsoft/Spotmarket-Switcher/dev/scripts/controller.sh)
Es gibt da auch mehr DEBUG Ausgaben.

EIn aufruf mir `DEBUG=1 bash ./controller.sh` Zeigt Dir dann mehr Infos und Hilft uns vielleicht weiter

Lg Chris

alex90alex, 2023-10-18:
mit der Version von gestern Abend funktioniert es noch nicht. Das aktivieren des Ladeplanes funktioniert, aber eben nicht umgekehrt. Wenn ich deine controller Version von oben verwende startet leider nichtmal der Ladeplan. Bei einem aktuellem Strompreis von 25 Cent und einer Preisgrenze von 29 Cent müsste doch der Ladeplan aktiviert werden oder? Beim Verwenden der Controller-Datei von gestern startet der Ladeplan.

Folgende Ausgabe erhalte ich:

Current price is 0.2553 EUR/kWh total price.
Lowest price will be 0.2211 EUR/kWh total price.
The average price will be 0.2817 EUR/kWh total price.
Highest price will be 0.3632 EUR/kWh total price.
Second lowest price will be 0.2407 EUR/kWh total price.
Third lowest price will be 0.248 EUR/kWh total price.
Fourth lowest price will be 0.2497 EUR/kWh total price.
Fifth lowest price will be 0.2553 EUR/kWh total price.
Sixth lowest price will be 0.2561 EUR/kWh total price.
D: condition_evaluation [ ( charge_at_second_lowest_price (0) == 1 && second_lowest_price_integer (2407000                                        0000000000) == current_price_integer (25530000000000004) ) evaluates to false ].
D: condition_evaluation [ ( charge_at_fourth_lowest_price (0) == 1 && fourth_lowest_price_integer (2497000                                        0000000000) == current_price_integer (25530000000000004) ) evaluates to false ].
D: condition_evaluation [ ( charge_at_solar_breakeven_logic (1) == 1 && feedin_price_integer (185000000000                                        00000) > current_price_integer (25530000000000004) + energy_fee_integer (15300000000000000) ) evaluates to                                         false ].
D: condition_evaluation [ ( charge_at_third_lowest_price (0) == 1 && third_lowest_price_integer (248000000                                        00000000) == current_price_integer (25530000000000004) ) evaluates to false ].
D: condition_evaluation [ ( charge_at_sixth_lowest_price (0) == 1 && sixth_lowest_price_integer (256100000                                        00000000) == current_price_integer (25530000000000004) ) evaluates to false ].
D: condition_evaluation [ ( charge_at_fifth_lowest_price (0) == 1 && fifth_lowest_price_integer (255300000                                        00000004) == current_price_integer (25530000000000004) ) evaluates to false ].
D: condition_evaluation [ ( use_start_stop_logic (1) == 1 && start_price_integer (29000000000000000) > cur                                        rent_price_integer (25530000000000004) ) evaluates to true ].
D: condition_evaluation [ ( switchablesockets_at_solar_breakeven_logic (0) == 1 && feedin_price_integer (1                                        8500000000000000) > current_price_integer (25530000000000004) + energy_fee_integer (15300000000000000) ) e                                        valuates to false ].
D: condition_evaluation [ ( switchablesockets_at_fifth_lowest_price (0) == 1 && fifth_lowest_price_integer                                         (25530000000000004) == current_price_integer (25530000000000004) ) evaluates to false ].
D: condition_evaluation [ ( switchablesockets_at_lowest_price (0) == 1 && lowest_price_integer (2211000000                                        0000000) == current_price_integer (25530000000000004) ) evaluates to false ].
D: condition_evaluation [ ( switchablesockets_at_third_lowest_price (0) == 1 && third_lowest_price_integer                                         (24800000000000000) == current_price_integer (25530000000000004) ) evaluates to false ].
D: condition_evaluation [ ( switchablesockets_at_sixth_lowest_price (0) == 1 && sixth_lowest_price_integer                                         (25610000000000000) == current_price_integer (25530000000000004) ) evaluates to false ].
D: condition_evaluation [ ( switchablesockets_at_second_lowest_price (0) == 1 && second_lowest_price_integ                                        er (24070000000000000) == current_price_integer (25530000000000004) ) evaluates to false ].
D: condition_evaluation [ ( switchablesockets_at_start_stop (0) == 1 && start_price_integer (2900000000000                                        0000) > current_price_integer (25530000000000004) ) evaluates to false ].
D: condition_evaluation [ ( switchablesockets_at_fourth_lowest_price (0) == 1 && fourth_lowest_price_integ                                        er (24970000000000000) == current_price_integer (25530000000000004) ) evaluates to false ].
D: is_charging_economical [ 1 - true ].
Victron scheduled charging is OFF. Battery SOC is at 58 %. Considering charging losses and costs, charging                                         is too expensive.
D: [ OK ]

alex90alex, 2023-10-18:
Wenn ich in Zeile 21 den economic check =0 setze, lädt das dev-Skript bei den oben genannten Bedingungen.

ckvsoft, 2023-10-18:
Ja, Bei economic check =0 wird der Check nicht durchgeführt. Somit wird auch ein davor gesetzte wahre Bedingung nicht revidiert.
Sonst könnte es sein das geladen werden darf aber der economic check sagt nein weil es doch zu teuer ist.

Sonst funktioniert alles wie gewollt? Oder braucht es noch testst? Hab nochmal ein Update gemacht. Bessere Lesbarkeit der debug Meldungen und ein paar kleinigkeiten gefixed.

ckvsoft, 2023-10-18:
> D: condition_evaluation [ ( use_start_stop_logic (1) == 1 && start_price_integer (29000000000000000) > cur rent_price_integer (25530000000000004) ) evaluates to true ].

Du meinst sicher
`D: condition_evaluation [ ( use_start_stop_logic (1) == 1 && start_price_integer (29000000000000000) > cur rent_price_integer (25530000000000004) ) evaluates to true ].`

Hier wird gesagt das geladen werden darf

Der economy check sagt aber dann Nein. Die Frage ist was soll er da machen?

Bei 2 Checkt er ob der Durchschnittspreis höher ist als die Kosten (Batterie Zeugs soweit ich das verstanden habe) Hier wird dann Nein gesagt wenn der Durchschnittspreis größer als die Kosten sind. Soll auch o sein Oder?

Bei 1 wird halt der höchste Preis genommen. Bis jetzt war  es so das der economy check meistens nein gesagt hat.

Ich hab noch ein Update in der Dev, da wird im Debug die Preis angezeigt die er vergleicht

ckvsoft, 2023-10-19:
Ich habe mir das Log von Heute nacht angesehen, es wurde kein einziges mal geladen
Ich versteh irgendwie den Sinn der Abfrage gar nicht. Weil

1.  **Der Höchstpreis ist wahrscheinlich immer höher als die Batterie Cost Berechnung**

2.  **Der Durchschnittpreis ist wahrscheinlich meistens höher als die Batterie Cost Berechnung**

Oder verstehe ich da was falsch?

ckvsoft, 2023-10-19:
Ok, mit Punkt zwei bin ich mal falsch gelegen ;)
Victron scheduled charging is ON. Battery SOC is at 100 %. Charging based on average price (0.243008 €) comparison makes sense. total_cost=0.3195 €

alex90alex, 2023-10-19:
> Sonst funktioniert alles wie gewollt? Oder braucht es noch testst? Hab nochmal ein Update gemacht. Bessere Lesbarkeit der debug Meldungen und ein paar kleinigkeiten gefixed.
>  Wie gesagt, der Ladeplan wird eben nicht beendet. 

Von den Funktionen des Spotmarketswitcher würde mir ja eine einfache Aktivierung und Deaktivierung ausreichen. Die Bedingung wäre: 1) Wenn der Strompreis der aktuellen Stunde < als meine Einspeisevergütung für Solarstrom ist = Ladeplan aktiv. 2) Wenn der Strompreis der aktuellen Stunde > als meine Einspeisevergütung für Solarstrom ist = Ladeplan inaktiv. (Insbesondere im Winter bzw. an windreichen Tagen ist der Strompreis glücklicherweise oft unter 19Cent/kWh.(brutto))

Möglicherweise bin ich auch zu blöd das Skript vernünftig zu installieren. Wenn ich dazu komme, könnte ich nochmal das Venus OS auf einer neuen SD Karte Installieren und testen ob es vielleicht am Venus OS liegt. Am sonst laufenden System von Victron möchte ich ungern weiter rumspielen.

christian1980nrw, 2023-10-19:
Beim Economic Check werden die höchsten oder alternativ durchschnittlichen Preise plus Ladeverluste (aktuelle kWh Preise mal Verlustprozente) plus Batteriekosten den aktuellen Preisen gegenübergestellt und abgebrochen wenn es zu teuer ist und man den Akku besser leer laufen lassen sollte. Denn wenn der Durschschnittspreis günstiger ist als der aktuelle Preis plus diese Kosten, dann ist das Laden Quatsch und man kann den Strom besser aus dem Netz beziehen. Das hat nichts mit der Einspeisevergütung zu tun.

ckvsoft, 2023-10-19:
> Beim Economic Check werden die höchsten oder alternativ durchschnittlichen Preise plus Ladeverluste (aktuelle kWh Preise mal Verlustprozente) plus Batteriekosten den aktuellen Preisen gegenübergestellt und abgebrochen wenn es zu teuer ist und man den Akku besser leer laufen lassen sollte. Denn wenn der Durschschnittspreis günstiger ist als der aktuelle Preis plus diese Kosten, dann ist das Laden Quatsch und man kann den Strom besser aus dem Netz beziehen. Das hat nichts mit der Einspeisevergütung zu tun, zumindest im Moment nicht. Verbesserungswürdig daran ist, das aktuell der höchste oder durchschnittliche Preis des gesamten Tages betrachtet wird. Da man den höchsten Preis ggf. aber zum jeweiligem Zeitpunkt schon hinter sich gelassen hat, wäre es besser wenn nur der höchste Preis und Durschnitt des restlichen Tages betrachtet würde.

Da muss ich mir mal einen älteren Code ansehen, vielleicht hab ich da was versehentlich weg rationalisiert.
So wie es jetzt ist macht es für mich irgendwie keinen sinn

ckvsoft, 2023-10-19:
> Von den Funktionen des Spotmarketswitcher würde mir ja eine einfache Aktivierung und Deaktivierung ausreichen. Die Bedingung wäre: 1) Wenn der Strompreis der aktuellen Stunde < als meine [...]

Ich nehme mal an Du hast tibber. Wo kommen da die Einspeise Preise her? Sind die Fix oder änderst Du das immer in der config.
Bei aWATTar bekomme ich monatlich die info über den aktuelle Preis. Ich möchte das natürlich nicht immer in der config umstellen müssen

christian1980nrw, 2023-10-19:
Die Einspeisepreise sind in der Regel die Vergütungen des Netzbetreibers für die PV-Einspeisung, die natürlich ohne Ladeverluste sind. Man müsste also den entgangenen Gewinn eigentlich auch noch mit einbeziehen in die Berechnungen.
@ckvsoft Schau mal ob Du aus dem historischem Code die Logik ableiten kannst, ich meine das hatte eigentlich schon funktioniert. Ansonsten denken wir es uns nochmal neu aus.

ckvsoft, 2023-10-19:
> Verbesserungswürdig daran ist, das aktuell der höchste oder durchschnittliche Preis des gesamten Tages betrachtet wird. Da man den höchsten Preis ggf. aber zum jeweiligem Zeitpunkt schon hinter sich [...]

Das könnte evtl für mich mehr Sinn machen. Weil egal was der Akku für kosten hat, möchte ich nicht auf alle fälle mit den höchsten Preis den Strom beziehen. 
Sollte man da nicht mit den aktuellen Preis gegen rechnen? Das würde für mich mehr sinn machen.

Jetzt schaut es ja so aus 
wenn der jetzige preis grade auch der günstigste ist dann darf der Ladeplan eigentlich Aktiv werden. Doch dann kommt der economy check der sagt das mein Akku mit allem drum und dran weniger kostet als der höchste Preis und schaltet den Ladeplan ab.

Beispiel: 
Höchstpreis 30.00 / kWh
Akkukosten + Alles  15.00 / kWh
Current Preis  2.00 / kWh

Da dürfte ja um 2.00 geladen werden doch es wird nicht weil die 30.00 (was ja schon waren oder irgendwann erst kommt) mehr ist als 20.00

ckvsoft, 2023-10-19:
> Die Einspeisepreise sind in der Regel die Vergütungen des Netzbetreibers für die PV-Einspeisung, die natürlich ohne Ladeverluste sind. Man müsste also den entgangenen Gewinn eigentlich auch noch mit [...]
@christian1980nrw 
Macht eigentlich das gleiche, nur funktionierte ja die charging_conditions Logic davor nicht. Die war ja so und so immer wahr.

Die economic_check abfrage wird ja auch ausgeführt und funktioniert, das sehe ich in den Debug ausgaben
Nur hat er die ganze Nacht nicht geladen weil eben bei einstellung 2 der durchschnittspreis höher als der errechnete Batteriekosten Preis war. Hat dadurch nicht eingeschaltet.

Ich habe also in der Nacht "billigen Strom verschenkt" und erst mit der neuen Preisliste  als es anscheinend billiger war doch relativ teuer geladen.

ckvsoft, 2023-10-19:
@christian1980nrw 
Eigentlich sollten wir die charging_conditions machen lassen und dann beim check es mit den gerade aktiven preis gegenrechnen.
Wenn die charging_conditions nicht anschlagen macht er ja den check so und so nicht weil ja execute_charging 0 ist und erst gar nicht in die abfrage gesprungen wird

ckvsoft, 2023-10-19:
@christian1980nrw 
> Die Einspeisepreise sind in der Regel die Vergütungen des Netzbetreibers für die PV-Einspeisung, die natürlich ohne Ladeverluste sind. Man müsste also den entgangenen Gewinn eigentlich auch noch mit einbeziehen in die Berechnungen. 

Ja nur bei aWattar sind die nicht fix. Man kann die zwar in einen Google Spreadsheet anschauen doch da müsste automatisch gehen.

christian1980nrw, 2023-10-20:
Nur damit wir nicht aneinander vorbei reden,
so wie im Excel war es ursprünglich mal gedacht. Siehe Anlage.
[](url)
[eco_check.xls](https://github.com/christian1980nrw/Spotmarket-Switcher/files/13048680/eco_check.xls)

christian1980nrw, 2023-10-20:
Nur damit wir nicht aneinander vorbei reden,
so wie im Excel war es ursprünglich mal gedacht. Siehe Anlage.

[eco_check.xls](https://github.com/christian1980nrw/Spotmarket-Switcher/files/13050894/eco_check.xls)

ckvsoft, 2023-10-20:
@christian1980nrw 
Ja so hätte ich es auch gemeint. Hab das evtl. etwas blöd ausgedrückt.
Nur jetzt macht die abfrage das genau umgekehrt.

Hab grad eine Abfrage gestartet

`Victron scheduled charging is ON. Battery SOC is at 100 %. Charging based on average price (0.27175 €) comparison makes sense. total_cost=0.5153 €`

Er ladet obwohl er ja nicht laden sollte, Die Gesamtkosten sind ja höher als der Durchschnittspreis

christian1980nrw, 2023-10-20:
@ckvsoft Moin. Habe das Excel vorhin nochmal getauscht, da war noch ein Fehler drin, war schon spät.
Wie rechnest Du denn die unterschiedlichen Preiseinheiten der APIs um oder lässt Du die in den jeweiligen Einheiten?

ckvsoft, 2023-10-20:
Ihr könnt gerne wieder die DVE Version testen. Sollte jetzt passen
`wget "https://raw.githubusercontent.com/ckvsoft/Spotmarket-Switcher/dev/scripts/controller.sh"
`

ckvsoft, 2023-10-20:
> @ckvsoft Moin. Habe das Excel vorhin nochmal getauscht, da war noch ein Fehler drin, war schon spät. 
@christian1980nrw 
Ja so gehts mir auch. Feile Stundenlang herum um einen Fehler einzugrenzen. Am nächsten Tag schau ich mir das an und denk mir was hab ich da für an Blödsinn gemacht. und schwupp Fehler gefunden ;-)

christian1980nrw, 2023-10-20:
@ckvsoft Übernimm bitte noch die beiden Änderungen. Habe an der Entso-E Api noch gefeilt.
Besorg Dir da mal einen API-Key (siehe Kommentare in der config.txt), Österreich müsste auch drin sein.
https://github.com/ckvsoft/Spotmarket-Switcher/compare/main...christian1980nrw:Spotmarket-Switcher:main

ckvsoft, 2023-10-20:
> @ckvsoft Übernimm bitte noch die beiden Änderungen. Habe an der Entso-E Api noch gefeilt. Besorg Dir da mal einen API-Key (siehe Kommentare in der config.txt), Österreich müsste auch drin sein. [ckvsoft/Spotmarket-Switcher@main...christian1980nrw:Spotmarket-Switcher:main](https://github.com/ckvsoft/Spotmarket-Switcher/compare/main...christian1980nrw:Spotmarket-Switcher:main)

hab ich gemacht.
Ich fang mit den Entso-E Zeugs nicht viel an ;-)

christian1980nrw, 2023-10-20:
> hab ich gemacht. Ich fang mit den Entso-E Zeugs nicht viel an ;-)

Falls aWATTar mal ne Störung hat, hättest Du noch ne Alternative. Du musst die API halt nur einmal kurz umkonfigurieren. Steht im Standard auf Deutschland.

ckvsoft, 2023-10-20:
Und die Preise sind dann ähnlich oder wie?

christian1980nrw, 2023-10-20:
Die Entso-E Preise sind nur in  EUR/MWh netto, ansonsten aber identisch. Ja, Failover der 3 API´s untereinander wäre ein tolles Feature. Entso-E bringt noch ganz viele andere Länder mit. Aber für Deutschland kann man alle 3 API´s aktivieren.  Für Österreich 2. Die Besonderheit bei Tibber wäre das Entso-E nur ein echtes Failover bei den Nettopreisen ist, wenn man Bruttopreise verwendet, müsste beim Failover zumindest eine Warnung kommen im Log.
Gruß! Christian

christian1980nrw, 2023-10-20:
@ckvsoft Wie war jetzt nochmal der letzte Stand mit dem Economic Check? Läuft er so wie in der Excel Tabelle? Ich müsste nämlich noch am Wochenende das Readme überarbeiten und dann ein neues Release erstellen wenn wir soweit sauber sind. Es sind jetzt wichtige Fixes im Master die auf jeden Fall in ein Release sollten, da ich das letzte zurückgezogen habe.

ckvsoft, 2023-10-20:
@christian1980nrw

Ja ich denke das es jetzt passt.

```
vorausgesetzt eines der conditions ist wahr dann kommt der check

wenn der check auf 0 eingestellt ist dann ladeplan ein

Höchstpreis kleiner als die kosten ladeplan ein
Höchstpreis größer als die kosten ladeplan aus

Durchschnitt-preis kleiner als die kosten ladeplan ein
Durchschnitt-preis größer als die kosten ladeplan aus

wenn keines der beiden zutrifft dann ladeplan aus

wenn keines der conditions wahr ist dann ladeplan aus

```

Alles Möglichkeiten habe ich noch nicht ausprobiert, aber es schaut mal Gut aus

ckvsoft, 2023-10-20:
Aber vor der Release sollte noch der installer angepasst werden, sodas die config nicht überschrieben wird

christian1980nrw, 2023-10-20:
Wir müssen die Logik dann auch noch mit der Einspeisevergütung verwursten (entgangener Gewinn bei Ladung in den Akku), sollen wir das vor dem Release noch tun? Vermutlich werden wir aber danach traurig sein wie selten der Charger anspringt.
Mit dem Installer achte ich drauf. Mal schauen wie ich das schaffe, da ich noch in anderen Projekten feststecke.

ckvsoft, 2023-10-20:
> Wir müssen die Logik dann auch noch mit der Einspeisevergütung verwursten (entgangener Gewinn bei Ladung in den Akku), sollen wir das vor dem Release noch tun? Vermutlich werden wir aber danach traurig sein wie selten der Charger anspringt. Wie man das am besten macht, muss ich auch nochmal drüber nachdenken, da man ja tagsüber mehr einspeisen kann wenn der Akku nachts vorgeladen wird. Mit dem Installer achte ich drauf. Mal schauen wie ich das schaffe, da ich noch in anderen Projekten feststecke.

Da bin ich mir nicht sicher ob das bis zum Wochenende :-) nicht ein Schnellschuss wird der nach hinten los geht.

Installer: ich hab die copy info aus der englisch beschreibung mal raus getan und im installer copier ich automatisch die sample.config.txt nach config.txt
mit der option -n
Mit dieser Option wird die Datei nicht überschrieben, falls sie bereits existiert. Es gibt keine Ausgabe oder Fehlermeldung, wenn die Datei nicht kopiert wird, der Befehl wird einfach stillschweigend beendet. Wenn die Ziel-Datei nicht existiert, wird sie wie gewohnt erstellt.

christian1980nrw, 2023-10-20:
Wie gehen wir denn mit zukünftigen Config Versionen um? Sollen wir die Config versionieren und eine Fehlermeldung ausgeben wenn sie nicht mehr supported ist?

ckvsoft, 2023-10-20:
Evtl. reicht es ja die Änderungen in die release Beschreibung zu schreiben
New, deprecated und removed
Allerdings müssten wir eine parser schreiben und dann darauf hinweisen das ein fehler in der config ist

alex90alex, 2023-10-20:
> Ihr könnt gerne wieder die DVE Version testen. Sollte jetzt passen `wget "https://raw.githubusercontent.com/ckvsoft/Spotmarket-Switcher/dev/scripts/controller.sh" `

ich habe eben mal das Skript geladen und getestet. 

Erster durchlauf mit folgenden Bedingungen: (Nach dem Ausführen des Skriptes wurde der Ladeplan gestartet)
stop_price=19.0    # stop above this price
start_price=19.0   # start below this price
feedin_price=18.5 # your feed-in-tariff of your solar system
energy_fee=15.3 # proofs of origin, allocations, duties and taxes (in case if stock price is at 0 Cent/kWh)
abort_price=19.0  # abort and never charge or switch if actual price is same or higher than this (Energy fees not included)

Den zweiten Durchlauf mit folgenden Bedingungen: (leider wurde hier wieder mal der **Ladeplan nicht gestoppt)**
stop_price=29.0    # stop above this price
start_price=29.0   # start below this price
feedin_price=18.5 # your feed-in-tariff of your solar system
energy_fee=15.3 # proofs of origin, allocations, duties and taxes (in case if stock price is at 0 Cent/kWh)
abort_price=29.0  # abort and never charge or switch if actual price is same or higher than this (Energy fees not included)

I: Current price is 0.258 EUR/kWh total price.
Lowest price will be 0.192 EUR/kWh total price.
The average price will be 0.268367 EUR/kWh total price.
Highest price will be 0.313 EUR/kWh total price.
Second lowest price will be 0.2065 EUR/kWh total price.
Third lowest price will be 0.2142 EUR/kWh total price.
Fourth lowest price will be 0.2255 EUR/kWh total price.
Fifth lowest price will be 0.2317 EUR/kWh total price.
Sixth lowest price will be 0.2403 EUR/kWh total price.
D: condition_evaluation [ ( charge_at_second_lowest_price (0) == 1 && second_lowest_price_integer (2065000        0000000000) == current_price_integer (25800000000000000) ) evaluates to false ].
D: condition_evaluation [ ( charge_at_fourth_lowest_price (0) == 1 && fourth_lowest_price_integer (2255000        0000000000) == current_price_integer (25800000000000000) ) evaluates to false ].
D: condition_evaluation [ ( charge_at_solar_breakeven_logic (1) == 1 && feedin_price_integer (185000000000        00000) > current_price_integer (25800000000000000) + energy_fee_integer (15300000000000000) ) evaluates to         false ].
D: condition_evaluation [ ( charge_at_third_lowest_price (0) == 1 && third_lowest_price_integer (214200000        00000000) == current_price_integer (25800000000000000) ) evaluates to false ].
D: condition_evaluation [ ( charge_at_sixth_lowest_price (0) == 1 && sixth_lowest_price_integer (240300000        00000000) == current_price_integer (25800000000000000) ) evaluates to false ].
D: condition_evaluation [ ( charge_at_fifth_lowest_price (0) == 1 && fifth_lowest_price_integer (231700000        00000000) == current_price_integer (25800000000000000) ) evaluates to false ].
D: condition_evaluation [ ( use_start_stop_logic (1) == 1 && start_price_integer (29000000000000000) > cur        rent_price_integer (25800000000000000) ) evaluates to true ].
D: condition_evaluation [ ( switchablesockets_at_solar_breakeven_logic (0) == 1 && feedin_price_integer (1        8500000000000000) > current_price_integer (25800000000000000) + energy_fee_integer (15300000000000000) ) e        valuates to false ].
D: condition_evaluation [ ( switchablesockets_at_fifth_lowest_price (0) == 1 && fifth_lowest_price_integer         (23170000000000000) == current_price_integer (25800000000000000) ) evaluates to false ].
D: condition_evaluation [ ( switchablesockets_at_lowest_price (0) == 1 && lowest_price_integer (1920000000        0000000) == current_price_integer (25800000000000000) ) evaluates to false ].
D: condition_evaluation [ ( switchablesockets_at_third_lowest_price (0) == 1 && third_lowest_price_integer         (21420000000000000) == current_price_integer (25800000000000000) ) evaluates to false ].
D: condition_evaluation [ ( switchablesockets_at_sixth_lowest_price (0) == 1 && sixth_lowest_price_integer         (24030000000000000) == current_price_integer (25800000000000000) ) evaluates to false ].
D: condition_evaluation [ ( switchablesockets_at_second_lowest_price (0) == 1 && second_lowest_price_integ        er (20650000000000000) == current_price_integer (25800000000000000) ) evaluates to false ].
D: condition_evaluation [ ( switchablesockets_at_start_stop (0) == 1 && start_price_integer (2900000000000        0000) > current_price_integer (25800000000000000) ) evaluates to false ].
D: condition_evaluation [ ( switchablesockets_at_fourth_lowest_price (0) == 1 && fourth_lowest_price_integ        er (22550000000000000) == current_price_integer (25800000000000000) ) evaluates to false ].
I: Victron scheduled charging is ON. Battery SOC is at 35 %. Charging based on condition met of: use_start        _stop_logic.
D: [ OK ]
root@raspberrypi4:/data/etc/Spotmarket-Switcher# DEBUG=1 bash ./controller.sh
I: Tibber today-data is up to date.
D: Variable: lowest_price | Original: 0.192 | Integer: 19200000000000000 | Len: 17
D: Variable: average_price | Original: 0.268367 | Integer: 26836700000000004 | Len: 17
D: Variable: highest_price | Original: 0.313 | Integer: 31300000000000000 | Len: 17
D: Variable: second_lowest_price | Original: 0.2065 | Integer: 20650000000000000 | Len: 17
D: Variable: third_lowest_price | Original: 0.2142 | Integer: 21420000000000000 | Len: 17
D: Variable: fourth_lowest_price | Original: 0.2255 | Integer: 22550000000000000 | Len: 17
[... cut ...]
Lowest price will be 0.192 EUR/kWh total price.
The average price will be 0.268367 EUR/kWh total price.
Highest price will be 0.313 EUR/kWh total price.
Second lowest price will be 0.2065 EUR/kWh total price.
Third lowest price will be 0.2142 EUR/kWh total price.
Fourth lowest price will be 0.2255 EUR/kWh total price.
Fifth lowest price will be 0.2317 EUR/kWh total price.
Sixth lowest price will be 0.2403 EUR/kWh total price.
D: Current price (0.2580€) is too high. Abort. (0.1900€)
root@raspberrypi4:/data/etc/Spotmarket-Switcher#

ckvsoft, 2023-10-20:
@alex90alex 
Gut das Du getestet hast. Ich hab den Fehler gefunden.
Bei abort wurde das script abgebrochen ohne den ladeplan zu stoppen.

Du kannst das nochmal versuchen, allerding muss ich das Morgen noch etwas anpassen.
Ich geh mal davon aus das die Shellys und Fritzln auch abgeschalten werden sollen

ckvsoft, 2023-10-20:
@christian1980nrw 
Die Frage für mich stellt sich warum hier überhaupt abgebrochen wird.
Evtl. Altbestand?

alex90alex, 2023-10-20:
gerne doch Danke euch für die Mühe. 

Einfach kurz schreiben wenn ich es wieder testen kann. Viele Grüße Alex

[10 more replies]

closed 2023-10-23 by christian1980nrw