butterwecksolutions/Spotmarket-Switcher #83: Batteriekosten Berücksichtigung

ckvsoft, 2023-10-16:
Kann man die Batteriekosten Berücksichtigung irgendwie deaktivieren?

Wenn die Kosten Tagelang über den Wert bleiben wird der Akku wohl nie geladen. Die Sonne reicht nicht immer dafür aus.

christian1980nrw, 2023-10-16:
Hi,
bei mir springt nachts aktuell auch nur die Warmwasser Wärmepumpe an, die Differenzen sind zu gering, da kann man gleich zu normalen Preisen ohne Ladeverluste tagsüber kaufen und den Akku leer lassen.
Du könntest die Batteriekosten und Ladeverluste heruntersetzen bis auf 0 wenn Du den Effekt verhindern willst. 
Ich habe aber vor, die Funktion in der Betrachtungsweise zu ändern. Es wird aktuell entweder der höchste oder durchschnittliche Tagespreis (vom gesamten Tag) verwendet. Ggf. hat man z.B. den höchsten Preis aber schon hinter sich gelassen, daher werde ich das wohl demnächst auf "den Rest des Tages" ändern. Wenn ich dann darin am ändern bin, mache ich die Batteriekosten dann auch per Schalter ganz abschaltbar. Mit dem SOC ist zwar eine gute Idee, allerdings soll das Script auch für andere Systeme als Victron universell nutzbar bleiben und wer weiß ob es sich da dann auslesen lässt. Daher wird er momentan nur zur Information mitgeloggt.

Gruß!
Christian

ckvsoft, 2023-10-16:
Danke, alles klar

Was mir nicht klar ist, ist folgender Code in Zeile 893

`if [[ $average_price_integer -ge $((current_price_integer+percent_of_current_price_integer+battery_lifecycle_costs_cent_per_kwh_integer)) ]]; then
    echo "I: Average price $average_price $Unit is lower than the sum of current price $current_price $Unit + ${energy_loss_percent}% energy loss for charging + battery lifecyle costs of $battery_lifecycle_costs_cent_per_kwh cent per KWh." | tee -a "$LOG_FILE"
`
Hier wird ja gecheckt ob der average_price_integer grösser oder gleich der berechnung von battery_lifecycle_costs ist
Wenn true wird aber dann geschrieben das laden sinn macht weil der preis ja kleiner ist.

Glaub das soll irgendwie nicht so sein. Oder?

lg Chris

christian1980nrw, 2023-10-16:
Eigentlich hatte ich die Logik auf Funktion getestet. Meinst Du da ist noch ein Fehler?
Ich habe sicherheitshalber Chatgpt dazu gefragt.

    Fall economic_check gleich 1:
        In diesem Fall wird geprüft, ob der höchste Preis (highest_price_integer) höher oder gleich dem Schwellenwert ist. Der Schwellenwert wird berechnet, indem der aktuelle Preis (current_price_integer), der Prozentsatz des aktuellen Preises (percent_of_current_price_integer) und die Kosten für die Lebensdauer der Batterie (battery_lifecycle_costs_cent_per_kwh_integer) addiert werden.
        Wenn der höchste Preis diesen Schwellenwert erreicht oder überschreitet, wird angenommen, dass das Laden wirtschaftlich sinnvoll ist.
        Entsprechende Meldungen werden protokolliert, und das Laden wird aktiviert, sofern use_victron_charger größer als 0 ist.

    Fall economic_check gleich 2:
        In diesem Fall wird die gleiche Prüfung durchgeführt wie im ersten Fall, jedoch unter Verwendung des durchschnittlichen Preises (average_price_integer) anstelle des höchsten Preises.
        Wenn der durchschnittliche Preis den Schwellenwert erreicht oder überschreitet, wird angenommen, dass das Laden wirtschaftlich sinnvoll ist.
        Wiederum werden entsprechende Meldungen protokolliert, und das Laden wird aktiviert, wenn use_victron_charger größer als 0 ist.

    Fall economic_check weder 1 noch 2:
        In diesem Fall wird angenommen, dass das Laden wirtschaftlich ist, unabhängig von den Preisen. Es wird angenommen, dass die Kosten für das Laden (Energieverlust und Batterielebensdauerkosten) im Verhältnis zu den Preisen nicht signifikant sind.
        Entsprechende Meldungen werden protokolliert, und das Laden wird aktiviert, sofern use_victron_charger größer als 0 ist.
        
        
Gruß!
Christian

ckvsoft, 2023-10-16:
So hätte ich das eh auch verstanden. Nur ist mir nicht klar warum laden wirtschaftlich ist wenn die preise höher oder gleich sind.

Sollte nicht geladen werden wenn es günstiger ist?

Lg Chris

ckvsoft, 2023-10-16:
`# Überprüfung, ob das Laden wirtschaftlich sinnvoll ist
if ((economic_check == 1)); then
    if [[ $highest_price_integer -ge $((current_price_integer + percent_of_current_price_integer + battery_lifecycle_costs_cent_per_kwh_integer)) ]]; then
        echo "I: Höchstpreis $highest_price $Unit ist höher als die Summe des aktuellen Preises $current_price $Unit + ${energy_loss_percent}% Energieverlust beim Laden + Batterielebenszykluskosten von $battery_lifecycle_costs_cent_per_kwh Cent pro KWh." | tee -a "$LOG_FILE"
        echo "   Laden ist sinnvoll." | tee -a "$LOG_FILE"
        if [0 -lt $use_victron_charger]; then
            $charger_command_turnon > /dev/null
            echo "I: Victron geplantes Laden ist EINGESCHALTET. Batterie SOC beträgt $SOC_percent %." | tee -a "$LOG_FILE"
        fi
    fi
elif ((economic_check == 2)); then
    if [[ $average_price_integer -ge $((current_price_integer + percent_of_current_price_integer + battery_lifecycle_costs_cent_per_kwh_integer)) ]]; then
        echo "I: Durchschnittspreis $average_price $Unit ist höher als die Summe des aktuellen Preises $current_price $Unit + ${energy_loss_percent}% Energieverlust beim Laden + Batterielebenszykluskosten von $battery_lifecycle_costs_cent_per_kwh Cent pro KWh." | tee -a "$LOG_FILE"
        echo "   Laden ist sinnvoll." | tee -a "$LOG_FILE"
        if [0 -lt $use_victron_charger]; then
            $charger_command_turnon > /dev/null
            echo "I: Victron geplantes Laden ist EINGESCHALTET. Batterie SOC beträgt $SOC_percent %." | tee -a "$LOG_FILE"
        fi
    else
        echo "I: Unter Berücksichtigung der Ladeverluste von ${energy_loss_percent}% + Batterielebenszykluskosten von $battery_lifecycle_costs_cent_per_kwh Cent pro KWh im Verhältnis zu den Einkaufspreisen ist der Ladevorgang zu teuer." | tee -a "$LOG_FILE"
        echo "   Laden ist nicht sinnvoll. Laden wird übersprungen." | tee -a "$LOG_FILE"
        echo "I: Victron geplantes Laden ist AUSGESCHALTET. Batterie SOC beträgt $SOC_percent %." | tee -a "$LOG_FILE"
        $charger_command_turnoff > /dev/null
    fi
fi
`
Etwas Formatiert ;-)

Hast Du einen anderen Code als der im Github?

Fall economic_check weder 1 noch 2: gibt es ja eigentlich nirgendwo. Oder übersehe ich da was?

lg Chris

christian1980nrw, 2023-10-16:
Das hat Chatgpt bei der Analyse festgestellt, kann aber Quatsch sein.
Ich habe leider gerade nicht wirklich Zeit mir das genau im Detail anzusehen, 
komme aber voraussichtlich im Laufe der Woche nochmal darauf zurück.

Wie gesagt, die letzte Version hatte ich eigentlich auf Funktion überprüft, vielleicht ist mir aber was entgangen,
ansonsten einfach mal den Schwellenwert händisch ausrechnen, Batteriekosten entsprechend setzen und bei niedrigen Preisen testweise mit unterschiedlichen Batteriekosten laufen lassen.

Gruß!
Christian

christian1980nrw, 2023-10-16:
@ckvsoft Waren die fehlerhaften charging conditions das Problem oder müssen wir hier noch weitermachen?
Danke Dir dafür. Ich muss mal die alten Tags durchgehen, wie lange das schon kaputt war.

ckvsoft, 2023-10-19:
Das kann man auch als erledigt ansehen. Deaktivieren kann man es ja jetzt. Das andere bin ich dran,

closed 2023-10-19 by ckvsoft