butterwecksolutions/Spotmarket-Switcher #96: Problem mit highest_price

ckvsoft, 2023-10-17:
Ich habe das Problem das mit Heute zb. der highest_price mit 9.80700 angezeigt wird.
Die Preise kommen aus aWattAr. Der tatsächliche highest_price ist aber 18.91300.

Ich konnte das Problem natürlich wieder auf AWK zurückführen. 9.80700 ist der höchste vor den komma einstellige Preis.
Lösche ich alle Preise die vor den Komma nur eine Stelle haben wird die highest_price richtig ermittelt.

Eine Lösung hätte ich mit  highest_price=$(grep -E '^[0-9]+\.[0-9]{5}$' "$file7" | tail -n1). grep und tail sollte ja auch im venusos installiert sein.

Ich hatte ja schon einige Probleme mit AWK Überläufen und hab deshalb bc verendet. BC gibts aber nicht am VenusOS. Nachinstallieren geht zwar aber viele wollen das nicht.

Mich nervt es aber das ich mich auf AWK nicht verlassen kann und da gibt es doch einiges.

Eine Bitte, kann mir hier jemand posten welche awk version im aktuellen VenusOS installiert ist. Ich habe die nächsten zwei Tage leider keinen Zugriff darauf. Auf meinen Test System ist AWK

mawk 1.3.4 20200120
Copyright 2008-2019,2020, Thomas E. Dickey
Copyright 1991-1996,2014, Michael D. Brennan

random-funcs:       srandom/random
regex-funcs:        internal
compiled limits:
sprintf buffer      8192
maximum-integer     2147483647

Ich schau mir mal die AWK aufrufe im Script an, kann aber nut die aWattAR sachen checken.

lg Chris

ckvsoft, 2023-10-17:
Mit
`awk '/^[0-9]+(\.[0-9]+)?$/ && $1 > max { max = $1 } END { print max }' /tmp/awattar_prices_sorted.txt`
Kann man ja mal schauen was raus kommt und ob das auch der höchste Preis ist

christian1980nrw, 2023-10-17:
Hi Christian,
also bei mir funktionierts, habe mal testweise auf aWATTar umgestellt. Nutze eine Version vom Spotmarket-Switcher von gestern (schon mit getrennter config.txt).
AWK Version folgt.

root@raspberrypi4:/data/etc/Spotmarket-Switcher# ./controller.sh
I: aWATTar today-data is outdated, fetching new data.
I: Please be patient. First we wait 17 seconds in case the system clock is not s                                                                                                             yncronized and not to overload the API.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  2946  100  2946    0     0  10751      0 --:--:-- --:--:-- --:--:-- 10751
I: Please verify correct system time and timezone:
   Tue Oct 17 18:18:23 CEST 2023

Current price is 16.59200 Cent/kWh net.
Lowest price will be 9.84000 Cent/kWh net.
The average price will be 13.0124 Cent/kWh net.
Highest price will be 19.10800 Cent/kWh net.
Second lowest price will be 10.00800 Cent/kWh net.
Third lowest price will be 10.44200 Cent/kWh net.
Fourth lowest price will be 10.55000 Cent/kWh net.
Fifth lowest price will be 10.79500 Cent/kWh net.
Sixth lowest price will be 10.83800 Cent/kWh net.
Victron scheduled charging is OFF. Battery SOC is at 98 %. Charging was not executed.
Turning OFF Fritz sockets.

christian1980nrw, 2023-10-17:
Hier nochmal mit aWATTar Österreich Preisen:

root@raspberrypi4:/data/etc/Spotmarket-Switcher# ./controller.sh
I: Fetching today-data data from aWATTar.
I: Please be patient. First we wait 13 seconds in case the system clock is not syncronized and not to overload the API.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  2950  100  2950    0     0   9076      0 --:--:-- --:--:-- --:--:--  9049
I: Please verify correct system time and timezone:
   Tue Oct 17 18:23:15 CEST 2023

Current price is 18.11900 Cent/kWh net.
Lowest price will be 10.40900 Cent/kWh net.
The average price will be 13.7373 Cent/kWh net.
Highest price will be 18.91300 Cent/kWh net.
Second lowest price will be 10.64200 Cent/kWh net.
Third lowest price will be 10.83600 Cent/kWh net.
Fourth lowest price will be 11.08400 Cent/kWh net.
Fifth lowest price will be 11.09300 Cent/kWh net.
Sixth lowest price will be 11.40400 Cent/kWh net.
Victron scheduled charging is OFF. Battery SOC is at 98 %. Charging was not executed.
Turning OFF Fritz sockets.

christian1980nrw, 2023-10-17:
Habe grad keinen Weg gefunden die Version anzuzeigen. Habe alle möglichen Parameter ausprobiert.
/usr/bin/awk
BusyBox v1.31.1 (2023-04-07 05:59:43 UTC) multi-call binary.

ckvsoft, 2023-10-17:
> Hier nochmal mit aWATTar Österreich Preisen:
>
> root@raspberrypi4:/data/etc/Spotmarket-Switcher# ./controller.sh I: Fetching today-data data from aWATTar. I: Please be patient. First we wait 13 [...]

Interessant

Current price is 18.11900 Cent/kWh net.
Lowest price will be 6.40000 Cent/kWh net.
The average price will be 11,8542 Cent/kWh net.
Highest price will be 9.80700 Cent/kWh net.
Second lowest price will be 7.00900 Cent/kWh net.
Third lowest price will be 7.29000 Cent/kWh net.
Fourth lowest price will be 7.69200 Cent/kWh net.
Fifth lowest price will be 8.22400 Cent/kWh net.
Sixth lowest price will be 9.19500 Cent/kWh net.

ckvsoft, 2023-10-17:
> Habe grad keinen Weg gefunden die Version anzuzeigen. Habe alle möglichen Parameter ausprobiert. /usr/bin/awk BusyBox v1.31.1 (2023-04-07 05:59:43 UTC) multi-call binary.

Ja das ist bei den awk varianten sehr unterschiedlich. Bei mir ist es awk -W version ;-)

Ich check das auf einen anderen System. Hab hier Ubuntu. Wie gesagt auf den Raspberry mit venusos hab ich erst wieder in zwei tagen zugriff.

christian1980nrw, 2023-10-17:
Falls Du auf Windows unterwegs bist, dann einfach wsl --install -d Debian und in wenigen Minuten hast Du eine shell.

Gruß!
Christian

ckvsoft, 2023-10-17:
> Falls Du auf Windows unterwegs bist, dann einfach wsl --install -d Debian und in wenigen Minuten hast Du eine shell.
> 
> Gruß! Christian

;-) Nein, ich hab seit 20 Jahre kein Windows mehr

ckvsoft, 2023-10-19:
Hat sich glaub ich erledigt. Den Effekt hab ich nur mit den awk von Ubuntu.
Debian, Manjaro und VenusOS funktionieren

closed 2023-10-19 by ckvsoft