butterwecksolutions/Spotmarket-Switcher #106: entsoe und include_second_day abort

ckvsoft, 2023-10-21:
Wenn bei Entsoe keine Preise für Morgen gibt wird abgebrochen und nicht wie bei aWATTar ohne diese Preise weiter gemacht.
Tibber kann ich nicht sagen wie sich der verhält, da waren Preise da.

Das schlechte bei so einen Abbruch ist das alles was läuft weiter läuft.

Vielleicht kannst Du das checken. Ich bin zwar schon Munter weil ich MotoGP schau komme aber wahrscheinlich erst wieder am Nachmittag dazu mir was anzuschauen. 

Ich habe sicherheitshalber einen älteren code getestet, der macht das auch. Nur um sicher zu gehen das ich da nix vermurkst habe als ich den Code umsortiert habe. (aktuelles dev)

lg Chris

ckvsoft, 2023-10-21:
I: File '/tmp/tibber_tomorrow_prices_sorted.txt' has no tomorrow data, we have to try it again until the new prices are online.

Also bei tibber geht es auch weiter. Ist also nur ein Fehler bei Entsoe

christian1980nrw, 2023-10-21:
> Wenn bei Entsoe keine Preise für Morgen gibt wird abgebrochen und nicht wie bei aWATTar ohne diese Preise weiter gemacht. Tibber kann ich nicht sagen wie sich der verhält, da waren Preise da.
>
> Das [...]

Ich habe nur nachmittags getestet, kann sein das mir das entgangen ist.

christian1980nrw, 2023-10-21:
Fehler behoben mit https://github.com/christian1980nrw/Spotmarket-Switcher/commit/c16944b79111607c8c0ea8c3f0e3b5c440da3846

closed 2023-10-21 by christian1980nrw