butterwecksolutions/Spotmarket-Switcher #115: discussion of logic regarding feed-in (export) tariff pre-charging

christian1980nrw, 2023-10-23:
Lets discuss the charging conditions in case of feeding in / exporting to the grid. @ckvsoft @alex90alex
If the prices and the total charging costs at night are lower than the export payout in the afternoon,
we should pre-charge the battery (maybe up to 100%).

Also if we are charging we should not forget the lost profit at charging.
It should be part of our calculations or am i wrong?

So that we don't talk past each other, we should carry out the example calculations in a shared spreadsheet first.
https://docs.google.com/spreadsheets/d/1Xkx3vjyCw-b7HOuuWABOCdY-kFDthEFuIPjaHk99Tdo/edit?usp=sharing

Regards
Christian

christian1980nrw, 2023-10-24:
@ckvsoft Meinst Du das reicht pauschal so wie im Excel oben aus oder sollten wir einen komplizierteren Ansatz wählen?

ckvsoft, 2023-10-24:
@christian1980nrw 
Ja ob das alles so Einfach ist kann ich auf die schnelle gar nicht sagen. Der Economy Check wird in diese Richtung schon passen.Evtl. müsste man einige Dinge ändern und anpassen.
Man müsste mal alles auf einen Nenner bringen wie cent/kw, alle Preise Netto oder Brutto, Einspeisevergütung ist ja nicht immer gleich denke ich mal, zumindest bei aWATTar nicht. Gibt es sonst noch variable kosten die berücksichtigt werden solle usw.

Die Frage ist reicht es in der Praxis so wie es ist oder Zahlt es sich aus sich meh anzunähern.

Wieviel die Sonne oder Heute liefert muss angepasst werden. Auch hier passt es oft zum Abfrage Zeitpunkt nicht mehr.
Zb. bei einer abfrage um 21 Uhr wird uns mitgeteilt des es Heute noch genug Sonne gibt und Morgen auch und ladet nicht. Sind die Wolken schon in die MJ eingerechnet oder nicht. Das alles ist offen.
Wie viel MJ brauche ich überhaupt um den Akku auf den gewünschten Stand zu bringen, Der SOC müsste mit eingerechnet werden.

Das sind alles noch Dinge die mir nicht klar sind. Das andere ist ob die Bash mit den vielen Rechenaufgaben nicht überfordert wird. Ich hab ja gesehen wie viel unterschied ein AWK machen kann. Dann noch die BASH auch hier gibt es unterschiede.
Am MacOS werden aus Lizenzgründen ältere Versionen der Bash installiert. Hier haben wir ja auch ein aktuelles problem das es beim macos kein declare -A gibt. Die standard shell beim mac ist seit macOS Catalina (Version 10.15) die zsh

Die Frage ist natürlich wie weit interessiert uns das. Stellen wir sicher das dieses Skript nur am VenusOS läuft? Doch da gibt es ja auch Unterschiede ob raspberry oder Victron-Geräte oder gar ein normaler PC.

lg Chris

christian1980nrw, 2023-10-24:
Also die Einspeisevergütung ist hier im Deutschland meist gleichbleibend über den Netzbetreiber vergütet (nicht über den Stromanbieter wie Tibber oder aWATTar) und nach Baujahr der Anlage für 20 Jahre festgelegt (vgl. https://www.sfv.de/lokal/mails/sj/verguetu ). Direktvermarktungsanlagen mit dynamischen Vergütungen gibt es in der Regel nur im Großanlagenbereich. Aber diese Anlagen sehe ich für das Script nicht als Zielgruppe.
Das Script sollte Betriebssystem unabhängig laufen, möglichst ohne zusätzliche Software. Vielleicht bekommen wir es gemeinsam mit @smoe hin, das z.B. auf einem Debian ohne Angabe eines Repositorys ein apt-get install Spotmarket-Switcher ausreicht.

smoe, 2023-10-24:
Hinsichtlich dynamischer Eispeisung möchte ich auf https://www.lumenaza.community/ verwiesen haben. Denke das kommt/ist schon da auch für kleinere Installationen. Werde dorthin wohl im Frühjahr wechseln - ist angeblich mit Tibber zu kombinieren.

Debian sehe ich erstmal nicht. Müsste dafür m.E. eher eine monolithische Anwendung sein/ein Service mit GUI/shell API, nicht so ein wiederholtes Anstarten via cron.

@ckvsoft hat offensichtlich ein paar wichtige Probleme identifiziert. Einmal angenommen wir wechseln zu Python - hinreichend ähnlich schwierig wie BASH für Neueinsteiger (die bislang nicht schlangestehen) - aber mit mehr Eleganz und Freiheiten bei der Programmierung. Aber hätte ich dafür gerade Valenzen übrig? Ne. Um festzustellen, was man eigentlich will, ist diese BASH-Lösung m.E. erstmal gar nicht so schlecht.

christian1980nrw, 2023-10-24:
Wir sollten erstmal bei der Bash bleiben und soweit optimieren bis wir alle einigermaßen mit dem Ergebnis zufrieden sind.  Ich sehe momentan keine allzu komplizierten Rechenaufgaben. Keep it simple, manchmal ist weniger mehr. 
Danach stünde dann die GUI auf der Agenda, und da Venus OS nativ NodeRed mitbringt, habe ich bereits eine mögliche Vorlage (die viel Entwicklungsarbeit spart) hier in Absprache mit dem Entwickler organisiert: https://github.com/christian1980nrw/Spotmarket-Switcher-GUI Wie wir das ganze dann verheiraten, müssen wir sehen, ist für mich auch Neuland.

Gruß! Christian

ckvsoft, 2023-10-24:
Die Gui brauchen wir wozu? Für die Ausgaben?
Das Script in ein Service zu wandeln sei ja nicht viel aufwand
In etwa so

```
trap 'stopservice' SIGTERM

stopservice() {
    echo "Das Skript wird beendet und führt die Bereinigung durch..."
    # code hier
}

while true; do
    # Hauptcode hier
    echo "Skript läuft..."

    # Ermitteln der aktuellen Minute und Sekunde
    current_min=$(date +'%M')
    current_sec=$(date +'%S')

    # Berechne, wie lange das Skript bis zur vollen Stunde schlafen sollte
    sleep_sec=$(( (60 - current_min) * 60 - current_sec ))

    sleep $sleep_sec
done

```

```
[Unit]
Description=Spotmarket-Switcher Service
After=network.target

[Service]
Type=simple
ExecStart=/pfad/zu/controller.sh
Restart=on-failure
User=benutzername
Group=gruppenname

[Install]
WantedBy=multi-user.target

```

ckvsoft, 2023-10-24:
evtl. bei einen neuen Major release darüber nachdenken. Mir reicht einstweilen das Logfile. Die Konfig werde ich ja auch nicht täglich ändern
Node-red würde für mich nicht in Betracht kommen. Verwende es auch jetzt nicht. Hab noch nicht den Vorteil entdecken können. ;-) Ist mir zu kompliziert

smoe, 2023-10-24:
Eine Status-Information würde noch benötigt - gerne aus den Logs - aber irgendetwas, das die Sensoren anfragen könnte, wäre möglicherweise beliebt.

smoe, 2023-10-24:
Victrons Infrastruktur wäre nur sekundär, richtig?

christian1980nrw, 2023-10-24:
> Victrons Infrastruktur wäre nur sekundär, richtig?

Ja, nachdem sie nur die Ideen des Spotmarket-Switchers übernommen und sich ansonsten für eine Eigenentwicklung in Kombi mit eigener Cloud entschieden haben, bleibt eine Unterstützung zwar, eine GUI kann aber auch unabhängig entwickelt werden. Ich würde Victron da nicht (mehr) zu sehr entgegenkommen wollen. NodeRed wäre da ein Kompromiss hatte ich gedacht, es kann aber auch was anderes werden.

Rellikon83, 2023-11-26:
Die Idee mit Node Red find ich gut. 

Das Problem an der controller.sh ist aktuell, dass man vor dem PC sitzen muss um zu checken was da gerade passiert. Über mobile devices oder remote ist da wenig machbar. Bei den Berechnungen steigt der Laie sowieso aus, bzw. will sich gar nicht damit beschäftigen. Im Grunde wollen die Endverbraucher, dass die Entscheidungen beim Script liegen. Grundsätzlich gibt es für die Leute zwei wesentliche Funktionen:

Verbrauchsoptimiert: Lade auf wenns billig ist und verbrauche wenn es teuer ist. -> Hier seid Ihr heute schon im teilweise nicht mehr leicht verständlichen Bereich bei den Prognosemodellen. Die sind aber notwendig wenn man das sinnvoll macht. NodeRed kann helfen die Komplexität abzubauen. Daher GUI ja bitte, aber erst wenn alles funktioniert.

FeedIn: Mach voll wenns billig ist, speise ein wenns teuer ist.

Btw. Super Job den Ihr macht. Ab 01.01. kann ich es hoffentlich auch testen.

smoe, 2023-11-26:
So richtig funktioniert das bei Victrons VRM mit der Dynamik noch nicht - jedenfalls nicht bei mir. Das kann aber auch daran liegen, dass ich aktuell für meinen grid-parallelen Betrieb  noch keinen Zähler bei Victron integriert habe. Kommt :)

Hinsichtlich Prognostic etc würde ich für dieses Projekt weitestgehend entspannt sein - was zu planen ist, wird man via der Tibber app bereits planen. Dort müsste man nur den Füllstand des aktuellen Stromspeichers noch mit anzeigen können.

christian1980nrw, 2023-11-26:
@Rellikon83 Dieses Projekt konzentriert sich aktuell ausschließlich auf kostengünstigen Bezug von Strom. Einspeisung aus dem Akku ins Netz ist aufgrund der Rahmenbedingungen in Deutschland mit EEG Anlagen nicht erlaubt. Man könnte ja sonst Strom aus unterschiedlichen Quellen z.B. Kohle miteinander vermischen so die Argumentation der Lobbypolitiker. Vielleicht ändert sich das irgendwann mal, aber so lange die Rahmenbedingungen hierzu weiter schlecht bleiben, wird diese Software auch nicht in die Richtung entwickelt.

Rellikon83, 2023-11-26:
@christian1980nrw unstrittig. Das gilt wie du schreibst für EEG Anlagen. Vermutlich der Großteil, daher ist das auch nicht der Fokus. 
Der Rest geht ja auch manuell.

Ich hatte heute das neue Update gezogen. 
Gut finde ich die Trennung zwischen Config und Controller. 

Gibt es ein Topic (hier oder Social Media) über die Funktionen die man so nutzt? Bspw. WP usw. Kommen sicherlich einige Ideen zusammen Dinge in die Nacht zu legen.

christian1980nrw, 2023-11-30:
@ckvsoft @smoe 
Ich bin aktuell nicht ganz zufrieden. Unser Ansatz is ja der, das wir zu günstigen Zeiten laden und dabei mit den durchschnittlichen Preisen vergleichen und bei zu hohen Ladeverlusten abbrechen. Ich nutze dazu die Gesamtpreise aus der Tibber-API. Leider reichen die Preisschwankungen meist nicht aus und es wird abgebrochen. Das hat im Herbst noch gut geklappt bei viel Wind im Netz parallel zur Sonne, momentan aber nicht mehr.

Es muss also ein anderer Ansatz her. Ich baue das Script bei mir gerade so um das zwar nach wie vor bei günstigen Preisen geladen wird, allerdings soll die Entladung nur noch bei höchsten Preisen erfolgen. Außerhalb dieser Zeiten wird das ESS abgeschaltet und aus dem Netz bezogen. Ich erhoffe mir davon eine deutlich höhere Rentabilität als bei der aktuellen Berechnung.

christian1980nrw, 2023-11-30:
@ckvsoft @smoe 

Ich habe mal eine neue experimentelle Version mit einer neuen Preismatrix bei mir im DEV hochgeladen.
Es gibt in der Controller.sh noch einiges an Optimierungsbedarf was eigentlich in Schleifen müsste, aber es geht ja erstmal um die obige Idee.

Folgendes ist außer Codeoptimierung noch to do:
- Eine zweite Preismatrix die die Next-Day Preise berücksichtigt sofern verfügbar (next day funktioniert damit also momentan nicht).
- Der Economic Check müsste so angepasst werden das für die Entscheidung ein Durchschnitt aus den aktivierten ESS-Preisen gebildet wird.

ckvsoft, 2023-12-01:
Ja am mac gibt es declare -A nicht. Zumindest nicht in der alten Bash Version vom Mac. Der ist ja vor einiger Zeit auf die ZSH umgestiegen.

Gibt es kein python am Cerbo GX? 
Ich hab ja mal nichts weitergemacht weil ich mir nicht sicher bin das ich das weiter mit der bash machen möchte.

christian1980nrw, 2023-12-01:
@ckvsoft 
...doch, der Cerbo kann Python3, aber naja, selbst wenn die Bash-Programmierung schmerzhaft ist,
finde ich das es sich lohnt das darauf zu lassen.
Es wird dadurch sichergestellt, das das auch auf ganz einfachen Endgeräten wie Routern läuft.
Und so kompliziert ist das Projekt ja jetzt auch nicht, irgendwann sind wir mit dem Code zufrieden
und dann wird da nicht mehr viel geändert.

ckvsoft, 2023-12-01:
Es müssen ja jetzt schon Klimmzüge gemacht werden um auf zwei unterschiedlichen Systemen zu laufen. Das Debuggen ist ebenfalls schwierig. Ich sehe keine Notwendigkeit das dieses Skript auf einen Router laufen sollte.  Wie man in der Vergangenheit gesehen hat funktioniert es ja nicht auf einer Standard Ubuntu installation. Da gibt es so viele Abhängigkeiten von den verwendeten Tools wie awk und co.  Ich habe Stunden unnötig versucht Fehler zu beheben bis ich drauf gekommen bin das der awk von Ubuntu Schuld war.
Ich kann zur Zeit aber auch nicht sagen was ich machen möchte,
Wie Du schon festgestellt hast ist die derzeitige Version nicht das richtige für die jetzige Jahreszeit.

ckvsoft, 2023-12-01:
Man müsste auch den SOC berücksichtigen. Das Problem ist das zur Zeit die Preise ja relativ hoch sind, mein Akku entleert wird und ich dann zu teuren Preisen Strom beziehe weil der Akku leer ist. Ein Statischer startpreis hilft hier auch nicht,
Ich schau mir mal das Skript an was neu ist.

ckvsoft, 2023-12-02:
> @ckvsoft @smoe
> 
> ```
> declare -A config_matrix24=(
>     ["Price"]="Charge Discharge Switch"
>     [1]="1 0 1"
>     [2]="1 0 1"
>     [3]="0 0 0"
>     [4]="0 0 0"
>     [5]="0 0 0"
>     [6]="0 0 0"
>     [7]="0 0 0"
>     [8]="0 0 0"
>     [9]="0 0 0"
>     [10]="0 0 0"
>     [11]="0 0 0"
>     [12]="0 0 0"
>     [13]="0 0 0"
>     [14]="0 0 0"
>     [15]="0 0 0"
>     [16]="0 0 0"
>     [17]="0 0 0"
>     [18]="0 0 0"
>     [19]="0 0 0"
>     [20]="0 1 0"
>     [21]="0 1 0"
>     [22]="0 1 0"
>     [23]="0 1 0"
>     [24]="0 1 0"
> )
> ```

Was ist bei dieser matrix was? 1 = Enabled?

christian1980nrw, 2023-12-02:
Ja genau. Abhängig von den Preisen 1-24 (niedrigster bis teuerster). 
Einmal für Ladevorgang, Entladevorgang (ESS) und Steckdosen.

ckvsoft, 2023-12-02:
Ok
Ich hab das etwas geändert damit es auch am MAC funktioniert.
Ist in meinen dev drinnen. Kannst es ja mal anschauen

christian1980nrw, 2023-12-02:
Ich schaue morgen mal, liege gerade mit Grippe im Bett.

christian1980nrw, 2023-12-03:
Du siehst es an ESS Status 7.
[image]

Außerdem kannst Du im ESS Menü nachsehen, ob das Limit auf 0W gesetzt wurde.
[image]

Ich bin noch mit Firmware v3.0~32 unterwegs. Liegt es vielleicht daran?

ckvsoft, 2023-12-04:
Ok, alles klar. Dachte ich sehe das irgendwie anders ;-)
Hab Version 3.12. 
Mein Akku war auf 10%, da schwankte es immer zwischen laden/entladen. Wär mir sonst nicht aufgefallen.

Wir sollten aber darüber nachdenken ob die Matrix nicht irgendwie dynamisch sein soll. In Verbindung mit den SOC.
Im Winter wenn auch die Wärmepumpe am start ist sind manchmal die biligsten x Preise die statisch eingestellt sind zu wenig um den Akku voll zu laden.

christian1980nrw, 2023-12-04:
Vielleicht ist greift bei Dir die Erhaltungsladung zu früh. Müsste man sich im Detail ansehen was auf Deinem Multiplus programmiert ist. 

Also in Verbindung mit dem SoC sehe ich das nicht vernünftig auswertbar. Wenn die Ladeverluste es zulassen, wird bei mir aktuell stur auf 85% aufgeladen mit den drei günstigsten Stunden. Man könnte aber eine monatsweise Dynamik oder eine abhängig vom Solarwetter einbauen.

ckvsoft, 2023-12-04:
Ja mit 3 Stunden komm ich nicht weit ;-)

christian1980nrw, 2023-12-05:
Vielleicht macht die Idee das dynamisch auf den SOC zu beziehen doch Sinn. Momentan hab ich nur 15% im Akku und die möchte ich dann natürlich nur zum teuersten Preis entladen. Ich baue das mal um in den nächsten Tagen.

ckvsoft, 2023-12-05:
Ja so ein Algorythmus wär nicht schlecht. Anhand des SOC die Anzahl von Stunden die benötigt werden um den Akku zu laden. Pro Stunde schaffe ich n% Ladung. Um den Akku auf x% zu laden werdeny Stunden benötigt. Das kann man ja dann auch lernen lassen.

ckvsoft, 2023-12-05:
Kann dann ja manchma sein das ich die 7 niedrigsten Preise brauch und manchmal nur 3

christian1980nrw, 2023-12-05:
Ich würde es eher als separat konfigurierbare Spalte in der Matrix einbauen... 
Dann kannst Du pro Preis was separates einstellen. Ich schaue mal.

christian1980nrw, 2023-12-05:
@ckvsoft Ich hab das mal umgesetzt und habe die sample.config und die controller.sh in meinem DEV aktualisiert.
Eigentlich war es ganz einfach. Siehe https://github.com/christian1980nrw/Spotmarket-Switcher/commit/691d5d79387222fe9e9c475f9ff855e152274570

ckvsoft, 2023-12-06:
Und warum? Hab das nicht aktiv. Hab dann natürlich Fehler

christian1980nrw, 2023-12-06:
Das Script schaltet erst beim nächsten Durchlauf wieder ab. Oder wenn zwei günstigste Stunden aufeinander folgen wird halt zweimal hintereinander eingeschaltet. Das ist so gewollt, da der Steckdosenstatus nirgends gespeichert ist. Ursprünglich hatte ich mal eine fast einstündige Pause im Script bis wieder abgeschaltet wurde, aber das hatte dann kurze Aussetzer zwischen zwei Laufzeiten zur Folge, also nicht so toll für Wärmepumpen. Die Alternative wäre den Status von der Steckdose auszulesen und abhängig davon eine Aktion auszuführen, das war mir aber zu kompliziert, da schalte ich lieber mehrfach ein oder mehrfach aus.

ckvsoft, 2023-12-06:
Dann sollte die abfrage vielleicht umgebaut werden. Wenn ich keine Steckdosen hab dann sollen die auch nicht versucht werden zu schalten

christian1980nrw, 2023-12-06:
Anso in der aktuellen Version im DEV kommt dann
W: skip Fritz DECT. not activated
W: skip Shelly Api. not activated

wenn man use_fritz_dect_sockets=0
und use_shelly_wlan_sockets=0
setzt.

Ggf. könnte man auf die Fehlermeldung verzichten, sie wird nervig sein für Nutzer die das nie planen zu nutzen.

christian1980nrw, 2023-12-06:
Hoffentlich scheint bald mal etwas die Sonne oder kommt eine größere Preisschwankung, bin gespannt ob die neue ESS-Preisdynamik wie erwartet funktioniert.

ckvsoft, 2023-12-06:
Ja egal. Zur Zeit hab ich andere Probleme mit laden und entladen. Das passt mir noch gar nicht.
Ich warte auf mein Akkuupdate und weiter Multiplus II. Vielleicht wirds dann besser.
Der Akku ist immer zum falschen Zeitpunkt leer bzw unter 20%
Heute gabs zb. 0kw von der PV und vom Netz wurde zu wenig lange geladen.

christian1980nrw, 2023-12-06:
Was hast Du denn für einen Akku? Ich habe damals meine Parameter aus Vorgabeempfehlungen vom Meintechblog konfiguriert.
Die passen soweit. Kommt er mal unter 10%, setzt direkt die Erhaltungsladung ein, auch bei langsamer Selbstentladung wird nach der ersten BMS-Warnung direkt wieder auf 10% aufgeladen. Moment, ich suche mal den Link raus wo die Werte standen.

ckvsoft, 2023-12-06:
Hat ja mit BMS jetzt nichts zu tun. Nur 17Kwh sind halt nicht gleich aufgeladen. Darum wollte ich ja das laden dynamisch eingestellt haben. Ich werde mir das überlegen wenn ich wieder Zeit dazu habe wie ich das lösen könnte

christian1980nrw, 2023-12-06:
https://meintechblog.de/2022/12/13/operation-hausspeicher-lifepo4-zellen-und-die-richtigen-spannungseinstellungen/

Em Ende des Posts ist eine Tabelle.

christian1980nrw, 2023-12-06:
> Hat ja mit BMS jetzt nichts zu tun. Nur 17Kwh sind halt nicht gleich aufgeladen. Darum wollte ich ja das laden dynamisch eingestellt haben. Ich werde mir das überlegen wenn ich wieder Zeit dazu habe wie ich das lösen könnte

Ach so, naja, da kannst Du dann von den günstigen Preisen möglichst viele aktivieren. Von welchem Parameter willst Du die Dynamik denn abhängig machen bzw. ableiten? Vom Solarwetter oder vom jeweils gültigem Monat? Das könnte man genauso wie es jetzt in meinem DEV beim ESS umgesetzt ist mit der Matrix tun.

ckvsoft, 2023-12-06:
Ja so genau weis ich das noch  nicht, sonst hätte ich ja schon was angefangen. Keine Ahnung, nur bin ich so wie es jetzt ist nicht wirklich zu Frieden. Mein Akku ist jetzt schon auf 12% im idle mode. Doch die teuersten Stunden hab ich Heute noch vor mir. Erst wieder um 23 Uhr gehts halbwegs.

christian1980nrw, 2023-12-06:
Dann besorg Dir doch im ersten Step mal einen API Key für das Solarwetter um vielleicht die Daten zumindest schonmal im Log aufzuzeichnen. Heute wäre also die Schwelle um zu 90% zu laden erreicht...

christian1980nrw, 2023-12-07:
@ckvsoft Vielleicht löst das Dein Problem und Du kannst mehr Preise aktivieren. Ich hab mal was gebaut für dynamisches SOC errechnet aus dem Solarwetter.
https://github.com/christian1980nrw/Spotmarket-Switcher/commit/00cbf5cbd710b52b4c8c180b198c59d67d67308d

alex90alex, 2023-12-08:
ich bin auch mal wieder dabei.. :-) 
Ich habe mir eben mal das aktuelle Skript geladen, bekomme aber folgende Fehlermeldung:

I: Bash Version: GNU bash, version 5.0.18(1)-release (arm-ve-linux-gnueabi)
I: Spotmarket-Switcher - Version 2.4.3-DEV
 [|]  Loading...Config validation passed.
I: aWATTar today-data is up to date.
I: Please be patient. A delay of 7 seconds will help avoid overloading the Solarweather-API.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100    42  100    42    0     0     76      0 --:--:-- --:--:-- --:--:--    76
date: invalid date '1970-01-01 '
date: invalid date '1970-01-01 '
controller.sh: line 541: ( - ) / 60: syntax error: operand expected (error token is ") / 60")
root@raspberrypi4:/data/etc/Spotmarket-Switcher#

Könnt ihr hier weiterhelfen? Habe ich etwas übersehen?
Beste Grüße
Alex

christian1980nrw, 2023-12-08:
Du hast das Solarwetter aktiviert und vermutlich keinen API Key eingetragen. Den musst Du Dir erstmal besorgen.

alex90alex, 2023-12-08:
Perfekt das wars wohl. 
Werden im aktuellen Skript noch die Preise von morgen mit einbezogen? Den entsprechenden Punkt habe ich in der Config mit =1 aktiviert. Die Ausgabe anbei bezieht sich aber wohl nur auf die Preise von heute.  (für eine verneinfachte Nachvollziehbarkeit der Berechnung habe ich aktuell den Verlust vom Victronsystem auf 10% gestellt.)
Zu meinem System: Victron MP2 5000 und ca. 16 kWh Speicher.
Falls Bedarf besteht, kann ich die kommenden 2-3 Wochen beim Testen vom Skript unterstützen. Programmieren kann ich leider nicht. ;-) 

I: Current price is 0.305 EUR/kWh total price.
I: Lowest price will be 0.2592 EUR/kWh total price.
I: The average price will be 0.287946 EUR/kWh total price.
I: Highest price will be 0.3059 EUR/kWh total price.
I: Second lowest price will be 0.2598 EUR/kWh total price.
I: Third lowest price will be 0.2615 EUR/kWh total price.
I: Fourth lowest price will be 0.2636 EUR/kWh total price.
I: Fifth lowest price will be 0.2681 EUR/kWh total price.
I: Sixth lowest price will be 0.2682 EUR/kWh total price.
I: Sunrise today will be 08:04 and sunset will be 16:27. Suntime will be 503 minutes.
I: Solarenergy today will be 0.8 megajoule per sqaremeter with 97.2 percent clouds.
I: Solarenergy tomorrow will be 3 megajoule per squaremeter with 92.1 percent clouds.
At 0.8 megajoule there will be a target-SoC of 90 set.
I: Victron scheduled charging is OFF. Battery SOC is at 17 %. Charging was not executed.
I: Victron discharging (ESS) is OFF. Battery SOC is at 17 %.  (total_cost=0.3355 â‚¬)
D: skip Fritz DECT. not activated
D: skip Shelly Api. not activated

alex90alex, 2023-12-08:
Okay alles klar. Ich stelle es wieder auf 20% und lasse das Skript mal übers Wochenende laufen.  Das Log-file wird aktuell nicht mehr befüllt oder? Ich würde gerne nachvollziehen, was das Skript jede Stunde berechnet/bewertet. :-)

alex90alex, 2023-12-10:
Aktuell scheint die Logik so für mein Gebäude/meine Hardware noch nicht zu passen. Mit den 3 günstigen Stunden heute Nacht wurde der Akku bereits so geladen, dass ich ohne Probleme das Haus über den ganzen Tag bis zum Abend/Nacht versorgen kann.  Durch das gesetzte Entladelimit von 0 wird aber gezielt das Entladen verhindert und lediglich auf die Abendstunden (21-24) gelegt. Die teuerste Stunde wäre heute jedoch von 18-19 Uhr. Möglicherweise hilft es, wenn ich das Entladelimit auf bspw. 500W erhöhe.

christian1980nrw, 2023-12-11:
Das hat mit Entladelimit nichts zu tun. 
Du musst die mittlere Spalte der Matrix anders konfigurieren wenn Du möchst das zu mehr Stunden entladen wird.

christian1980nrw, 2023-12-15:
@alex90alex @ckvsoft Ich habe im DEV gerade eine neue Version 2.4.4-DEV hochgeladen, bei der die Preise des Folgetages wieder funktionieren sollten. In der Config gibt es für die 48 Stunden nun eiine zweite Matrix die verwendet wird sobald diese am Nachmittag vorliegen. Im Code gibt es einige Verbesserungen, ob alles wie bisher funktioniert, müsste noch getestet werden.

christian1980nrw, 2023-12-15:
...also ich hab schon was gefunden. Wenn die jeweiligen Preise erreicht werden, wird die jeweilige Aktion nicht ausgelöst, z.B. eingeschaltet oder geladen. Ich schaue heute Abend mal woran es liegt, kann nur eine Kleinigkeit sein.

christian1980nrw, 2023-12-15:
[image]

So, die aktuelle DEV Version sollte nun etwas besser funktionieren.
Die Ausgabe enthält nun auch Angaben zum dynamischen SOC-Ladeziel (abhängig vom Solarertrag) und den dynamischen Entladepreisen (abhängig vom Akkufüllstand).

christian1980nrw, 2023-12-18:
...ich mache mal hier zu, da kein Feedback mehr kommt.

closed 2023-12-18 by christian1980nrw

ckvsoft, 2023-12-23:
Ja sorry, hatte keine Zeit zum testetn.
Mir ist nur aufgefallen das bei mir die start logic nicht funktioniert hat. Gestern zb. hätte den ganzen Tag geladen werden dürfen.

christian1980nrw, 2023-12-30:
@ckvsoft Du hast Recht, habe es heute mal ausprobiert und einen neuen Issue aufgemacht.
Ich bin leider schon seit Wochen krank, daher läuft aktuell recht wenig an diesem Projekt.

ckvsoft, 2023-12-31:
Ja ich komm ja einstweilen auch zu nix. 
Ich würde es so sehen das für mich der Start Preis vorang hat und nicht der economic check. Bei Preisen der letzten Tagen hat der gar keinen Sinn. Zu den Preisen darf er den ganzen Tag vom Netz leben und den Akku voll halten

christian1980nrw, 2023-12-31:
Ich nutze die Bruttopreise aus der Tibber App, da ist der Check schon realistischer, da die Steuern und Abgaben die Schwankungen erhöhen. Ich überlege die anderen APIs auch optional auf Bruttopreise umzustellen. Bei mir steht der Check außerdem auf highest_price. Leider sind die Preisschwankungen tagsüber oft zu klein, so das sich das Laden des Akkus nur selten lohnt.

[6 more replies]