butterwecksolutions/Spotmarket-Switcher #125: Script will fail if Entso-E has an API Error

christian1980nrw, 2023-10-24:
Output:
root@raspberrypi4:/data/etc/Spotmarket-Switcher# DEBUG=1 sh controller.sh
I: Bash Version: GNU bash, version 5.0.18(1)-release (arm-ve-linux-gnueabi)
I: Spotmarket-Switcher - Version 2.3.7
 [\]  Loading...Config validation passed.
I: Fetching today-data data from Entsoe.
D: No delay of download of entsoe data since DEBUG variable set.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   941  100   941    0     0   1297      0 --:--:-- --:--:-- --:--:--  1296
D: No delay of download of entsoe data since DEBUG variable set.
D: No delay of download of entsoe data since DEBUG variable set.
cp: can't stat '/tmp/entsoe_today_prices_sorted.txt': No such file or directory
root@raspberrypi4:/data/etc/Spotmarket-Switcher#

XML-Content of entsoe_today_prices.xml (not our fault) in case of API Error:
`<?xml version="1.0" encoding="UTF-8"?>
<Acknowledgement_MarketDocument
	xmlns="urn:iec62325.351:tc57wg16:451-1:acknowledgementdocument:7:0">
	<mRID>3639ea49-2e45-4</mRID>
	<createdDateTime>2023-10-24T20:35:35Z</createdDateTime>

	<sender_MarketParticipant.mRID codingScheme="A01">10X1001A1001A450</sender_MarketParticipant.mRID>
	<sender_MarketParticipant.marketRole.type>A32</sender_MarketParticipant.marketRole.type>

	<receiver_MarketParticipant.mRID codingScheme="A01">10X1001A1001A450</receiver_MarketParticipant.mRID>
	<receiver_MarketParticipant.marketRole.type>A39</receiver_MarketParticipant.marketRole.type>

	
	<received_MarketDocument.createdDateTime>2023-10-24T20:35:35Z</received_MarketDocument.createdDateTime>

	<Reason>
		<code>999</code>
		<text>Unexpected exception, please contact support and provide exception ID.
Exception ID: 18b69966-0e7a-42f5-95a6-593a889981a4</text>
	</Reason>

</Acknowledgement_MarketDocument>`

christian1980nrw, 2023-10-24:
@ckvsoft Even if error code 999 is part of the XML, awk will not find it. I tested it before and i will try out an older version of the controller.sh.

christian1980nrw, 2023-10-24:
`download_entsoe_prices() {
    local url="$1"
    local file="$2"
    local output_file="$3"
    local sleep_time="$4"

    if [ -z "$DEBUG" ]; then
        echo "I: Please be patient. First we wait $sleep_time seconds in case the system clock is not syncronized and not to overload the API."
        sleep "$sleep_time"
    else
        echo "D: No delay of download of entsoe data since DEBUG variable set." >&2
    fi

    if ! curl "$url" >"$file"; then
        log_info "E: Retrieval of entsoe data from '$url' into file '$file' failed."
        exit 1
    fi

    if ! test -f "$file"; then
        log_info "E: Could not find file '$file' with entsoe price data. Curl itself reported success."
        exit 1
    fi

    if [ -n "$DEBUG" ]; then echo "D: Entsoe file '$file' with price data downloaded" >&2; fi

    if [ ! -s "$file" ]; then
        log_info "E: Entsoe file '$file' is empty, please check your entsoe API Key."
        exit 1
    fi

    if [ -n "$DEBUG" ]; then echo "D: Entsoe file '$file' with price data downloaded"; fi

    awk '
/<Period>/ {
    capture_period = 1
}
/<\/Period>/ {
    capture_period = 0
}
capture_period && /<resolution>PT60M<\/resolution>/ {
    valid_period = 1
}
valid_period && /<price.amount>/ {
    gsub("<price.amount>", "", $0)
    gsub("<\/price.amount>", "", $0)
    gsub(/^[\t ]+|[\t ]+$/, "", $0)
    prices = prices $0 ORS
}
valid_period && /<\/Period>/ {
    exit
}

/<Reason>/ {
    in_reason = 1
    error_message = ""
}

in_reason && /<code>/ {
    gsub(/<code>|<\/code>/, "")
    gsub(/^[\t ]+|[\t ]+$/, "", $0)
    error_code = $0
}

in_reason && /<text>/ {
    gsub(/<text>|<\/text>/, "")
	gsub(/^[\t ]+|[\t ]+$/, "", $0)
    error_message = $0
}

/<\/Reason>/ {
    in_reason = 0
}

END {
    if (error_code == 999) {
        print "E: Entsoe data retrieval error:", error_message
        exit 1
    } else if (prices != "") {
        printf "%s", prices > "'"$output_file"'"
    } else {	
	if ("'"$output_file"'" != "'"$file13"'") {
            print "E: No prices found in the today XML data."
			exit 1
        }
    } 
            print "E: No prices found in the tomorrow XML data."
}
' "$file"

    if [ -f "$output_file" ]; then
        sort -g "$output_file" > "${output_file%.*}_sorted.${output_file##*.}"
        timestamp=$(TZ=$TZ date +%d)
        echo "date_now_day: $timestamp" >> "$output_file"
    fi

    # Check if tomorrow file contains next day prices
    if [ "$include_second_day" = 1 ] && grep -q "PT60M" "$file" && [ "$(wc -l <"$output_file")" -gt 3 ]; then
        cat $file10 >$file8
        #    echo >> $file8
        if [ -f "$file13" ]; then
            cat "$file13" >>"$file8"
        fi
        sed -i '25d 50d' "$file8"
        sort -g "$file8" >"$file19"
        timestamp=$(TZ=$TZ date +%d)
        echo "date_now_day: $timestamp" >>"$file8"
    else
        cp $file11 $file19 # If no second day, copy sorted price file.
    fi
}`

This old version is working and reading out the API error.

christian1980nrw, 2023-10-24:
This will work (contains still some old code) @ckvsoft :
https://github.com/christian1980nrw/Spotmarket-Switcher/commit/cce154e40e60da882db54916925e6f0f8ca141d8

ckvsoft, 2023-10-24:
@christian1980nrw 

I can't remember changing anything else with the ensoe stuff other than replacing exit 1.

But are you still going to adjust it? Or

christian1980nrw, 2023-10-24:
Yes, i can fix it and tell you after. I hope the API will still stay broken the next few minutes.

christian1980nrw, 2023-10-24:
Should be fixed with my last commits (last one was https://github.com/christian1980nrw/Spotmarket-Switcher/commit/cce3a19b65947875a533e0c11385e6e59dfe1211 ) @ckvsoft

closed 2023-10-24 by christian1980nrw

ckvsoft, 2023-10-25:
log_info ist kein command sondern eine funktion im script
Die Fehler beim mac war ja nur weil die funktion zu früh im code aufgerufen wurde.
Ist mit der umreihung der functionen behoben worden.

```
log_message() {
  message="$1"
  if [ -z "$(command -v log_info)" ]; then
    logger -p user.info -t "Spotmarket-Switcher" "$message"
  else
    log_info "$message"
  fi
}
```
kann ich wieder entfernen, so wird ja log_info nie ausgeführt ;-)