butterwecksolutions/Spotmarket-Switcher #131: main.yml should start the installer of the current branch

christian1980nrw, 2023-10-24:
I dont know why my changes are not working.
I wand to submit the correct branch to the installer.
See https://github.com/christian1980nrw/Spotmarket-Switcher/commit/da2cd0726c0b36a33ef4fe64f267ba9bf9442e1b

ckvsoft, 2023-10-25:
I guess I overlooked that too.
The actor should actually also be specified.

```
- name: Install
  run: ./victron-venus-os-install.sh
  env:
    DEBUG: 1
    DESTDIR: /tmp/testing
    BRANCH: ${{ steps.set-branch-name.outputs.branch_name }}
    ACTOR: ${{ github.actor }}

```
Otherwise the wrong controller.sh will be tested.
Of course, the installer also needs to be customized

ckvsoft, 2023-10-25:
This also applies to venus.yml. That would also be wrong

christian1980nrw, 2023-10-25:
@ckvsoft Can you maybe fix that if you know how? My head is already paining from debugging this.

ckvsoft, 2023-10-25:
@christian1980nrw 
Yes, me too ;-)
I usually program in c++, so bash etc. always confuse me.

I'll try to adjust that

christian1980nrw, 2023-10-26:
@ckvsoft Crazy this github stuff, right? 
Will the installer download the dev or main branch or the branch of the pull-request?

ckvsoft, 2023-10-26:
> @ckvsoft Crazy this github stuff, right? Will the installer download the dev or main branch or the branch of the pull-request?

@christian1980nrw 

the current one - main or dev, yours or mine. depending on who the source was.
Unfortunately, Merged cannot be tested because it is only virtual and cannot be accessed via wget from the installer

ckvsoft, 2023-10-26:
@christian1980nrw 
you may have to rewrite it
That the installer gets the current one from the current branch via wget. But the controller script is executed by merged. that should work.
You can pass the sample.config.txt as an additional env

ckvsoft, 2023-10-26:
erledigt

christian1980nrw, 2023-10-26:
@ckvsoft 
Thank you, this was really painful to debug, i know.

christian1980nrw, 2023-10-26:
@ckvsoft We have one last problem, if you pull dev into main, the versioning script fails.
Run # Attempt to remove the -DEV suffix from the version number
On branch main
Your branch is up to date with 'origin/main'.

nothing to commit, working tree clean
Error: Process completed with exit code 1.

ckvsoft, 2023-10-26:
> @ckvsoft We have one last problem, if you pull dev into main, the versioning script fails. Run # Attempt to remove the -DEV suffix from the version number On branch main Your branch is up to date with 'origin/main'.
> 
> nothing to commit, working tree clean Error: Process completed with exit code 1.

sollte jetzt funktionieren ;-)

closed 2023-10-27 by christian1980nrw