butterwecksolutions/Spotmarket-Switcher #140: Behavior for reserving capacity when the sun is shining should be improved

christian1980nrw, 2023-10-26:
We are already reading sunrise and sunset.
I: Sunrise today will be 08:10 and sunset will be 18:13. Suntime will be 603 minutes.

If the running time of the script is later as sunset, it makes no sense to stop charging because of solar weather.
There is no need to reserve battery capacity anymore for charging by sun, there will be no sun anymore.
In this case script runtime at 19:00 and later that day should not be canceled anymore.

Karkas66, 2024-01-09:
I think there is overall improvement potential in the solar calculation, today for example the Script wasted a lot of solar Power for me, because battery idled at 100% for an hour but the script turned of ESS discharging. 
I dont know if this is relevant:
`Jan  9 10:00:43 einstein cron.info CROND[13790]: (root) CMDOUT (^[[1;32mI: Sunrise today will be 08:27 and sunset will be 16:26. Suntime will be 479 minutes.^[[0m)
Jan  9 10:00:43 einstein cron.info CROND[13790]: (root) CMDOUT (^[[1;32mI: Solarenergy today will be 4 megajoule per sqaremeter with 4 percent clouds. The temperature is -7.3°C with 1.1cm snowdepth.^[[0m)
Jan  9 10:00:43 einstein cron.info CROND[13790]: (root) CMDOUT (^[[1;32mI: Solarenergy tomorrow will be 4.4 megajoule per squaremeter with 4.4 percent clouds. The temperature will be -6.3°C with 1.1cm snowdepth.^[[0m)
Jan  9 10:00:44 einstein cron.info CROND[13790]: (root) CMDOUT (^[[1;32mI: There is snow on the solar panels at negative degrees. Target SOC will be set to 90% (max value of the matrix).^[[0m)
Jan  9 10:00:46 einstein cron.info CROND[13790]: (root) CMDOUT (^[[1;32mI: Victron scheduled charging is OFF. Charging was not executed. (Total charging costs=0.4790 €)^[[0m)
Jan  9 10:00:47 einstein cron.info CROND[13790]: (root) CMDOUT (^[[1;32mI: Victron discharging (ESS) is OFF. Battery SOC is at 81%.^[[0m)`

Your script disabled discharging and on this sunny day the 19% battery went straight into the battery, charging it to 100% in minutes. Then the system essentialy disabled the solar panels for the rest of the hour, wasting all that energy until full hour:

```
Jan  9 11:00:53 einstein cron.info CROND[27819]: (root) CMDOUT (^[[1;32mI: Solarenergy today will be 3.6 megajoule per sqaremeter with 3.6 percent clouds. The temperature is -7.3°C with 1.1cm snowdepth.^[[0m)
Jan  9 11:00:53 einstein cron.info CROND[27819]: (root) CMDOUT (^[[1;32mI: Solarenergy tomorrow will be 4.4 megajoule per squaremeter with 4.4 percent clouds. The temperature will be -6.3°C with 1.1cm snowdepth.^[[0m)
Jan  9 11:00:53 einstein cron.info CROND[27819]: (root) CMDOUT (^[[1;32mI: There is snow on the solar panels at negative degrees. Target SOC will be set to 90% (max value of the matrix).^[[0m)
Jan  9 11:00:55 einstein cron.info CROND[27819]: (root) CMDOUT (^[[1;32mI: Victron scheduled charging is OFF. Charging was not executed. (Total charging costs=0.4708 €)^[[0m)
Jan  9 11:00:56 einstein cron.info CROND[27819]: (root) CMDOUT (^[[1;32mI: Victron discharging (ESS) is ON. Battery SOC is at 100%.^[[0m)
```

Script noticed "oh shit battery full" lets reenable ESS discharging...

I don´t know actually how to improve this but this bugs me somehow, even if it is only a few cents wasted

Karkas66, 2024-01-09:
What I´m trying to say is that this sentence is just wrong:

`There is snow on the solar panels at negative degrees. Target SOC will be set to 90% (max value of the matrix).`

christian1980nrw, 2024-01-09:
Hi @Karkas66 ,
> Script noticed "oh shit battery full" lets reenable ESS discharging...
ESS disabling and enabling can be adjusted by the middle colum of the 24 & 48 price matrix.
I think for your enviroment you should configure it a bit dirrerent so that ESS re-enables earlier and more price-hours get enabled at lower SOC.

Please check also your settings of
use_solarweather_api_to_abort
abort_solar_yield_today
and abort_solar_yield_tomorrow

I will check in the evening if there is a possible issue so that ESS is not getting re-enabled in case of a abort.

> There is snow on the solar panels at negative degrees. Target SOC will be set to 90% (max value of the matrix).
This is a new feature. You can modify line 1373 of the controller.sh from 1 to 5 or more centimeters if this was not true and already melted. I will move this parameter to the config file soon.
if awk -v temp="$temp_today" -v snow="$snow_today" 'BEGIN { exit !(temp < 0 && snow > 1) }'; then

Regards
Christian

Karkas66, 2024-01-09:
Nice, thanks. Will do all that

Karkas66, 2024-01-09:
This looks way better:
```I: Please verify correct system time and timezone:
   Tue Jan  9 16:38:37 CET 2024
I: Current price is 0.3427 EUR/kWh total price.
I: The average price will be 0.322167 EUR/kWh total price.
I: Highest price will be 0.3616 EUR/kWh total price.
I: Sorted prices: 1:0.2907 2:0.2917 3:0.2925 4:0.293 5:0.2932 6:0.2937 7:0.2938 8:0.2941 9:0.2954 10:0.2965 11:0.2968 12:0.2968
                  13:0.3048 14:0.3061 15:0.3098 16:0.3105 17:0.3122 18:0.3149 19:0.3162 20:0.3164 21:0.3169 22:0.3197 23:0.3201 24:0.3204
                  25:0.3211 26:0.3225 27:0.3234 28:0.3249 29:0.3252 30:0.3316 31:0.3326 32:0.3333 33:0.3336 34:0.3344 35:0.3379 36:0.3382
                  37:0.3408 38:0.3427 39:0.3457 40:0.3457 41:0.3478 42:0.3491 43:0.3508 44:0.3545 45:0.3546 46:0.3571 47:0.3587 48:0.3616

I: Charge at prices: 1 2 3 4 5 6
I: Dynamic ESS discharge (depending SOC) at prices: 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47 48
I: Sunrise today will be 08:27 and sunset will be 16:26. Suntime will be 479 minutes.
I: Solarenergy today will be 1.9 megajoule per sqaremeter with 1.9 percent clouds. The temperature is -7.2°C with 0.7cm snowdepth.
I: Solarenergy tomorrow will be 4.4 megajoule per squaremeter with 4.4 percent clouds. The temperature will be -6.8°C with 0.5cm snowdepth.
I: At 1.9 megajoule there will be a dynamic SOC charge-target of 71% calculated. The rest is reserved for solar.
I: Victron scheduled charging is OFF. Charging was not executed. Total charging costs: 0.5114€
I: Victron discharging (ESS) is ON. Battery SOC is at 71%.
```

christian1980nrw, 2024-01-10:
@Karkas66 
Please upgrade your controller.sh to v2.4.12.
ESS was not turned on again in case of a abort because of enough solarenergy.
I fixed this.

Regards Christian

Karkas66, 2024-01-10:
Did it!
`...
I: Charge at prices: 1 2 3
I: Dynamic ESS discharge (depending SOC) at prices: 14 15 16 17 18 19 20 21 22 23 24
I: Sunrise today will be 08:27 and sunset will be 16:28. Suntime will be 481 minutes.
I: Solarenergy today will be 4.3 megajoule per sqaremeter with 4.3 percent clouds. The temperature is -7.4°C with 1.2cm snowdepth.
I: Solarenergy tomorrow will be 3.8 megajoule per squaremeter with 3.8 percent clouds. The temperature will be -5.1°C with 1.1cm snowdepth.
I: There is snow on the solar panels (snowdepth > 1cm) at negative degrees. Target SOC will be set to 85% (max value of the matrix).
D: Before evaluating charging conditions - execute_charging: 0
D: Before evaluating discharging conditions - execute_discharging: 0
D: Before evaluating switchable sockets conditions - execute_switchablesockets_on: 0
D: After evaluating charging conditions - execute_charging: 0
D: After evaluating discharging conditions - execute_discharging: 1
D: After evaluating switchable sockets conditions - execute_switchablesockets_on: 0
I: Victron scheduled charging is OFF. Charging was not executed. Total charging costs: 0.4977€
I: Victron discharging (ESS) is ON. Battery SOC is at 71%.
`
I personaly think that 2 cm snow on a full black pannel on a sunny day like yesterday and today is not a problem. The panel will heat up enough to let the snow melt and slide down. I have full snow free panels for all of the week. 
I will change the value of the snow again to 3cm in controler.sh