butterwecksolutions/Spotmarket-Switcher #159: start stop locic

ckvsoft, 2023-11-06:
Ich verwende jetzt ein paar Tage die Start Stop logic.
Aber ich kann nicht feststellen was diese eigentlich genau machen sollte.

Es wird ja nur überprüft ob der startpreis grösser als der currentpreis ist. Was sollte der Stoppreis tun?

christian1980nrw, 2023-11-06:
Der Parameter ist noch Überrest einer anfänglichen Idee mit festen Preisgrenzen zur Scriptlaufzeit zu arbeiten.
Werden sie unter- oder überschritten wird eine Aktion ausgelöst.
Sollen wir eine neue config Version veröffentlichen oder das erst bei der nächsten größeren Änderung tun?
Ich habe z.B. vor vom Hersteller Sonnen die Sonnebatterie zu integrieren, dan wird sowieso eine neue Version notwendig sein.

ckvsoft, 2023-11-06:
OK. Zur Zeit wird ja nur mal gecheckt ob der Preis unterhalb des Startpreises liegt. Das ist für mich eh OK, nur für den Stopppreis müsste ja mehr gemacht werden.
Wird gerade geladen und der Preis ist noch unterm Stopppreis dann lade weiter oder so ähnlich. Sehe aber wenig Sinn darin. Könnte ja den Startpreis höher stellen und der Stopppreis hat sich erledigt.

Sonnen und Sonnenbatterie sagt mir jetzt mal gar nichts.

christian1980nrw, 2023-11-06:
Ich entferne den Parameter mit der nächsten Version aus der Config. Ein Preis reicht zur Abgrenzung ob geladen wird oder nicht.

christian1980nrw, 2023-11-26:
Fixed with https://github.com/christian1980nrw/Spotmarket-Switcher/pull/160 . Please test the new version at main & dev.

closed 2023-11-26 by christian1980nrw

ckvsoft, 2024-01-08:
Zur INFO
Nachdem ich immer noch Probleme mit Bash und Debuggen habe habe ich genug.
Habe begonnen mit Python3 ähnliches zu erarbeiten. Dabei bin ich drauf gekommen das man die ESS über mqtt Steuern kann. Somit ist es nicht zwingend notwendig es direkt am Venus zu laufen.

christian1980nrw, 2024-01-09:
@ckvsoft 
Schade, hat bisher Spaß gemacht mit Dir. Ja es macht nur wenig Freude das auf der Bash umzusetzen, ich finde aber das mittlerweile trotzdem ein gutes Ergebnis dabei rumgekommen ist. Ich werde auf jeden Fall noch die Themen bearbeiten die schon unter "issues" beschrieben sind und anschließend neben Victron als zweites System Ende Frühjahr das von der Firma Sonnen (Sonnebatterie) integrieren. Dann vielleicht noch eine GUI. Danach gehen mir so langsam die Ideen aus was man noch verbessern könnte. Wirst Du Dein Script von der Logik her anders gestalten? Gabs einen bestimmten Knackpunkt der auf der Bash nicht geklappt hat? Du meintest ja neulich das wir unterschiedliche Ansichten haben wann geladen werden soll, aber das Zeitfenster der veröffentlichten Preise ist nunmal maximal auf den Folgetag beschränkt, da gibts nicht so viel Spielraum. 
Ich berücksichtige übrigens jetzt auch die Schneetiefe. Ab 1cm wird fällt die Solarreservierung der Kapazität weg...
Kannnst Du mir noch Infos geben wie das per mqtt zu steuern ist? Dann kann ich das auch noch darauf umbauen, würde Vorteile für die Leute bringen, die das auf einem Router laufen lassen wollen.

Gruß! Christian

ckvsoft, 2024-01-09:
Ja, leider. Bis zu einen gewissen Zeitpunkt hats mich gepackt. Doch der Ärger über das Fehlersuchen ohne Debug und Einzelschritt Möglichkeit hat mich dann zum aufgeben gezwungen.
Ich finde das Tool ja unbeschreiblich Gut, auch so wie es jetzt schon ist. 

Das es die Preise nur bis max. nächsten Tag gibt das stört mich gar nicht so. Da hab ich mich eh nie durchgeblickt was da genau passiert ;-)
Ja ideen hätte ich schon noch, nur die Zeit wohl weniger ;-) Hast Du Dir für die Solarberechnung schon mal forecast.solar angeschaut? Da kommen recht brauchbare Werte raus. Da kannst angeben wie viel watt deine Panels habe, deinen Standpunkt, wohin die Panels ausgerichtet sin und in welchen Winkel die montiert sind. Dann bekommst werte die eigentlich recht nah an de tatsächlichen erreichten werten waren. Habe das jetzt aber erst drei Tage verfolgt.

Du hast nicht Lust auf Python :-) Ich habe als c++ entwickler auch lange gebraucht mich damit anzufreunden, doch jetzt gefällt mir das recht Gut. Als IDE verwende ich PyCharme.

MQTT, Du musst auf deinen Venus mqtt aktivieren. Dann kannst sowohl im lokalen netz als auch über die VRM darauf zugreifen.

[image]

Mit einen MQTT Explorer darauf zugreifen (http://mqtt-explorer.com/)
Dann sieht man unter N alle Einträge
Wenn Du nichts siehst kann es sein das Du eine message absetzen musst 
`R/die_serien_numme_vom_victronr/system/0/Serial` damit wird es angestossen