butterwecksolutions/Spotmarket-Switcher #170: Victron ESS Batterylife on, oder without Batterylife

kLOsk, 2023-12-19:
Hi Christian,
Kannst du sagen ob es Sinn macht Batterylife auszuschalten, bei der Nutzung von Spotmarket Switcher, oder besser anlassen? Bei mir hängt der SOC bei 80% fest, da ich  noch Probleme mit den richtigen Einstellungen vom Spotmarket Switcher habe.
Ich werde später auch nochmal nen Thread aufmachen bzgl config.txt Fragen :D

Beste Grüße
Daniel

christian1980nrw, 2023-12-19:
Hi Daniel,
es steht bei mir auf "Optimized without battery Life". 
Ich hatte irgendwo auch mal gelesen das man das für Lifepo4 nicht benötigt.
Ja das kann sein das Du sonst Probleme bekommst da das die Steuerungslogiken ggf. nicht zulässt.
Gruß! Christian

closed 2023-12-19 by christian1980nrw

christian1980nrw, 2023-12-19:
BattyLife sorgt dafür das der Akku voll bleibt bei schlechtem Wetter für z.B. TopBalancing.
Wenn der Akku lange nur im unteren Bereich genutzt wird, driften die Zellen irgendwann auseinander.
Die Funktion macht aber keinen Sinn wenn Du nachts günstig laden willst.
Du solltest aber prüfen wie tief Du ihn entladen kannst und den  Erhaltungsspannungswert passend setzen so das die Batterie immer wieder auf 10 oder 11% zurück geladen wird z.B. bei Selbstentladung durch die Elektronik.
Ich habe mich mit meinen Einstellungen hieran orientiert:
https://meintechblog.de/2022/12/13/operation-hausspeicher-lifepo4-zellen-und-die-richtigen-spannungseinstellungen/

Gruß!
Christian

kLOsk, 2023-12-19:
Danke für den Link. Ich habe Batterylife jetzt auch mal deaktiviert und die Anlage hat die Batterie aus 79% runtergesaugt. Jetzt ist der ESS State #7 (User configured a discharge limit of zero) und es wird nicht mehr entladen. Minimum SOC ist 10%.
Kennst Du das Problem?

Meine config.txt ist:

economic_check=1                             
disable_inverting_while_only_switching=1 
limit_inverter_power_after_enabling=-1  

Ich denke das ist der Grund: disable_inverting_while_only_switching=1  ?

Beste Grüße
Daniel

christian1980nrw, 2023-12-19:
Hi, nein das ist Absicht und wird in der 24er und 48er Matrix festgelegt (mittlere Spalte, SOC).
Es wird erst auf teurere Preise gewartet bis tiefer entladen wird, außer Du konfigurierst die Matrix anders.

Gruß!
Christian