butterwecksolutions/Spotmarket-Switcher #171: Questions about config.txt

kLOsk, 2023-12-19:
Hi Christian,
Can you help me make more sense of the individual settings in config.txt?

* **economic_check**
    * If this is set to 0, when will the battery charge?
    * Can you give an example when 1 an 2 should be chosen?
* **disable_inverting_while_only_switching**
    * Can you explain this? What does "if switching is turned on" mean?
* **limit_inverter_power_after_enabling**
    * I'm also lost with this setting. Can you explain what it means?
* **start_price**
    * How does this relate to the settings above. Does it overrule it at any times and if for example set to -5, charging will never start if the price is not below -5?
* **use_start_stop_logic** and **switchablesockets_at_start_stop** and **charge_at_solar_breakeven_logic** and **switchablesockets_at_solar_breakeven_logic**
    * How do these settings relate to the settings above?
* config_matrix_target_soc_weather
    * Can you provide a good resource to calculate how many megajoule are needed for a full day of power?

Also, regarding the Victron settings, I feel we should add soem more information to the documentation. My guess in Winter most users would be happy to simply fully charge their bateries full at any given time of the day. So the scheduler needs to be set to 0:00 - 23:59 and SOC Limit off, right?
Generally, I believe this would be the right setting now, since SOC limit and best charging time is take care of by the matrix now, right?

Thanks
Daniel

christian1980nrw, 2023-12-19:
I will reply tomorrow to your topics...

Regards Christian

christian1980nrw, 2023-12-20:
> Hi Christian, Can you help me make more sense of the individual settings in config.txt?
> 
>     * **economic_check**
>       
>       * If this is set to 0, when will the battery charge?
>       * Can you give an example when 1 an 2 should be chosen?
this is to calculate if charging is too expensive (energy price + charging losses + lifecycle costs)
it will be compared with:
1=highest price 2=average price

> 
>     * **disable_inverting_while_only_switching**
>       
>       * Can you explain this? What does "if switching is turned on" mean?
> 
If you activate a switch, maybe ESS is turned on. To avoid that at this time the battery will be used, you can use this switch. During switching ESS will be turned off so that the battery stays full.

>     * **limit_inverter_power_after_enabling**
>       * I'm also lost with this setting. Can you explain what it means?
> 
Maybe you are using a user defined setting at the ess menu at "maximum inverter power". This setting will be zero sometimes because the script is disabling ESS with this. Your user defined setting will get lost in this case. This setting is to return to your user defined setting (only if you are using this) after re-enabling ess.

>     * **start_price**
>       
>       * How does this relate to the settings above. Does it overrule it at any times and if for example set to -5, charging will never start if the price is not below -5?
no need to use this setting. its only active if you are using use_start_stop_logic=1. Maybe somtimes there are negative prices and in this case you can set a price to start charging independent from the other logics. This is a old and confusing feature that can be removed in future versions.

> 
>     * **use_start_stop_logic** and **switchablesockets_at_start_stop** and **charge_at_solar_breakeven_logic** and **switchablesockets_at_solar_breakeven_logic**
>       
>       * How do these settings relate to the settings above?

use_start_stop_logic=1  
switchablesockets_at_start_stop=1 
this two settings belong together

charge_at_solar_breakeven_logic=1
switchablesockets_at_solar_breakeven_logic=1

    use_start_stop_logic:
    When set to 1, this option activates a logic to start or stop charging (or other actions) based on a specified start price. It likely means the system begins charging when the energy price drops below a certain threshold and stops when the price exceeds that threshold.

    switchablesockets_at_start_stop:
    If enabled (set to 1), this option allows for additional devices (like a water heater) to be turned on or off using AVM Fritz DECT200/210 switchable sockets, depending on the start/stop logic. So, if the start/stop conditions are met, extra devices can be automatically switched on or off.

    charge_at_solar_breakeven_logic:
    This option, when enabled, controls charging based on comparing the current energy costs including fees with the feed-in tariff of your solar system. If the energy costs are lower than the feed-in tariff, the system starts charging. This optimizes the use of the solar system by charging when it's more cost-effective to consume energy rather than feeding it into the grid.

    switchablesockets_at_solar_breakeven_logic:
    Similar to the start/stop logic, when set to 1, this option enables the switching on or off of additional devices via switchable sockets based on the solar breakeven logic. Devices are turned on when it’s more economical to use self-generated energy and turned off when it’s better to feed energy into the grid.

> 
>     * **config_matrix_target_soc_weather**
>       
>       * Can you provide a good resource to calculate how many megajoule are needed for a full day of power?
> 
You can view historic settings of your place with this link (replace Dortmund with your place) and choose a date of your choice.
You should compare it with your own solar system. Maybe choose old dates where it was full.
https://weather.visualcrossing.com/VisualCrossingWebServices/rest/services/timeline/Dortmund%20Germany/2023-07-01/2023-07-31?unitGroup=metric&elements=solarenergy&include=days&key=YOURAPIKEY&contentType=csv

> 
> Also, regarding the Victron settings, I feel we should add soem more information to the documentation. My guess in Winter most users would be happy to simply fully charge their bateries full at any given time of the day. So the scheduler needs to be set to 0:00 - 23:59 and SOC Limit off, right? 
No sometimes the battery will stay empty because of the charging losses its too expensive to charge. Its cheaper to buy energy from the grid if the price differences during the days are too low.
If you activate the solar weather api, the charging limit gets calculated automatically dependig how much solar energy is expacted that day.
Generally, I believe this would be the right setting now, since SOC limit and best charging time is take care of by the matrix now, right?
Its a bit tricky to find out the best settings for your system but i am sure you will find them.
> 
> Thanks Daniel

Regards Christian

closed 2023-12-20 by christian1980nrw

kLOsk, 2023-12-20:
Thanks for your responses. I have some follow up questions tho. I'm planning on rewriting the config and docs to make this more understandable so please excuse my annoying questions here :)

> this is to calculate if charging is too expensive (energy price + charging losses + lifecycle costs) it will be compared with: 1=highest price 2=average price
What happens if this is set to 0?
Can you explain when a user should use the highest price for comparison and when he should use the average price. Also I assume this is also very much depending on the new matrix feature and battry capacity. So depending on total battery capacity and average daily consumption of the system it might be more beneficial to charge below average or below highest price. My assumption would be for most users with a battery size of half their daily consumption the average price would be the right choice?
> If you activate a switch, maybe ESS is turned on. To avoid that at this time the battery will be used, you can use this switch. During switching ESS will be turned off so that the battery stays full.
So this is relating to shelly / fritz switches, correct. If no Shelly / Fritz switches are enabled, this doesn't do anything?  I haven't really looked into switches yet as I don't have use for them yet. Can you explain when the switches are currently tunred on at all and how this relates to this setting then?
> Maybe you are using a user defined setting at the ess menu at "maximum inverter power". This setting will be zero sometimes because the script is disabling ESS with this. Your user defined setting [...]
I see, so this is referring to this here: https://www.victronenergy.com/media/pg/Energy_Storage_System/en/configuration.html#UUID-46d07eb6-7e0a-d2f8-20f2-a8b6eed61b1d
So if Limit Inverter Power in Victron Settings is turned off, this does nothing.
> no need to use this setting. its only active if you are using use_start_stop_logic=1. Maybe somtimes there are negative prices and in this case you can set a price to start charging independent from [...]
Understood
> use_start_stop_logic=1 switchablesockets_at_start_stop=1 this two settings belong together
>
> charge_at_solar_breakeven_logic=1 switchablesockets_at_solar_breakeven_logic=1
>
> ```
> use_start_stop_logic: [...]
Understood. What would "other actions" mean tho? If I understand this correctly the other actions would be switches and those are defined separately below?
> switchablesockets_at_start_stop:
> If enabled (set to 1), this option allows for additional devices (like a water heater) to be turned on or off using AVM Fritz DECT200/210 switchable sockets, [...]
Understood.
> charge_at_solar_breakeven_logic:
> This option, when enabled, controls charging based on comparing the current energy costs including fees with the feed-in tariff of your solar system. If the energy [...]
understood. This is independent of use_start_stop_logic ?
> switchablesockets_at_solar_breakeven_logic:
> Similar to the start/stop logic, when set to 1, this option enables the switching on or off of additional devices via switchable sockets based on the [...]
understood
> You can view historic settings of your place with this link (replace Dortmund with your place) and choose a date of your choice. You should compare it with your own solar system. Maybe choose old [...]
About the matrix: Is the average user supposed to change the values there or can they be assumed to be a good general standard setting. "1 95 1" and "0 95 0" Can you explain what this means. 95 is SOC but what are  1 and 0 doing?

Thanks
> Regards Christian

christian1980nrw, 2023-12-20:
> Thanks for your responses. I have some follow up questions tho. I'm planning on rewriting the config and docs to make this more understandable so please excuse my annoying questions here :)
>
>
> What [...]

If 0 there is no check if charging is economic and it will be charged depending the prices.
No i dont recommend to use average because in this case you maybe loose money. Buying from grid is maybe cheaper as using from the pre-charged battery.
If you have a small battery you should find good SOC settings at the matrix so that discharging will be stopped at cheap prices.

> So this is relating to shelly / fritz switches, correct. If no Shelly / Fritz switches are enabled, this doesn't do anything? I haven't really looked into switches yet as I don't have use for them [...]

Yes its useful if you use switchable sockets and charger at the same time.

> I see, so this is referring to this here: https://www.victronenergy.com/media/pg/Energy_Storage_System/en/configuration.html#UUID-46d07eb6-7e0a-d2f8-20f2-a8b6eed61b1d So if Limit Inverter Power in [...]

This parameter is not needed in most cases. If not set the script will switch from off to 0w and from 0w to off again.

> Understood
>
>
> Understood. What would "other actions" mean tho? If I understand this correctly the other actions would be switches and those are defined separately below?

Correct. This switch is to deactivate this logic for charging AND switching.

> Understood.
>
>
> understood. This is independent of use_start_stop_logic ?

Yes. In case if the grid buy energy is cheaper than your grid sell energy, you are able to charge independent of the matrix with this. 

> understood
>
>
> About the matrix: Is the average user supposed to change the values there or can they be assumed to be a good general standard setting. "1 95 1" and "0 95 0" Can you explain what this [...]

95 means i this price is the current price, it will only discharge if the battery is almost full. The first 1 will activate charging at this price line and the second 1 will activate switching. Currently all switchable sockets will be activated at the same time.

I am still testing the settings of the matrix because this feature is new. So i currently dont know if the default settings are good if there will be more sun again.

> 
> Thanks
> 
> > Regards Christian