butterwecksolutions/Spotmarket-Switcher #172: CMDOUT (/data/etc/Spotmarket-Switcher/controller.sh: eval: line 1242: syntax error near unexpected token `(')

kLOsk, 2023-12-26:
Saw this syntax error last night in var/log/messages
Possibly description the ( parentheses need to be escaped?

` local conditions=("${!1}") local conditions=\("${!1}"\)`

` local descriptions=("${!2}") local descriptions=\("${!2}"\)`

Can anyone confirm?

Thanks
Daniel

ckvsoft, 2024-01-02:
I can't confirm it. On which system are you using the script? Which bash version?

christian1980nrw, 2024-01-02:
Please try it again with the latest version and close the issue if it cannot be reproduced.

Thanks
Christian

christian1980nrw, 2024-01-03:
@kLOsk @ckvsoft 
Important fix of this issue: https://github.com/christian1980nrw/Spotmarket-Switcher/commit/1182cf2c9956653ae7bec51a8c95022179563435
Today i was wondering why my switchable socket was not turned on and something of the log was missing.
I remembered this issue, and set my clock back to the time when it should be turned on and bingo:
./controller.sh: eval: line 1308: syntax error near unexpected token `('

After the fix everything will work again. 
Sorry for this mistake, there were many changes migrating from 6 lowest prices to 24/48.

Regards
Christian

closed 2024-01-03 by christian1980nrw