butterwecksolutions/Spotmarket-Switcher #175: Prices of the past should be forgotten if second day is activated

christian1980nrw, 2023-12-31:
ToDo

bacanol, 2024-01-08:
Hi @christian1980nrw, there is another issue with pricing and the second day. Case if you switch from two day scope to only one day, the script incorporates the second day until the price files rotate. Maybe you can check if the setting was switched or the file contains second day pricing and remove the entries. Thank you and best regards, Nico

christian1980nrw, 2024-01-08:
Hi @bacanol, you mean if you change the configuration? Which API are you using? Best practice not to run into an untested issue it to delete the downloaded files at /tmp after every reconfiguration. Regards Christian

christian1980nrw, 2024-01-08:
@bacanol In version 2.4.11, I have added a new feature. If the configuration file or the controller script is modified within the last 60 minutes, a cleanup of the /tmp directory will be performed during the next execution of the program, except for the logfile.

butterwecksolutions, 2025-03-01:
Implemented with update https://github.com/butterwecksolutions/Spotmarket-Switcher/commit/ee7653c91894a0813fa42f881b7630a2f59534dd (Beta)

closed 2025-03-01 by butterwecksolutions