butterwecksolutions/Spotmarket-Switcher #181: Victron ESS #7 limit inverter power blocks system during sunny days

bacanol, 2024-01-29:
Hello @christian1980nrw,

I am operating 4 Victron Energy instances with your script and Tibber and it is working quite well. Now with the current version installed, I discovered when the system is operating towards a nearly full battery the ESS #7 limit inverter power reduces the power of the DC-DC system as it is not allowed to provide power to the loads in the AC side. I guess the "bug" is kind of happening because of the nature of ESS #7 limit inverter power. Furthermore the energy is not sold to the public grid in case the battery is full and the sun in shining, which is at least better than to not use the energy.

I do understand your motivation on this one and it is working great if the battery is not (nearly) full and the day is not sunny. Maybe there is a way to circumvent this issue. 

Here you can see the issue with one of the systems (I was able to reproduce this on another system as well):

[image]

The battery became full charged approx 12h and 12:15 I disabled manually the ESS #7.

christian1980nrw, 2024-01-29:
@bacanol 
Currently there are 2 methods to discharge earlier during sunny days:
1. use of abort_solar_yield_today and abort_solar_yield_tomorrow so that the script will be disabled
2. use of a more aggressive discharge matrix so that it ill discharge earlier (please compare your matrix with the sample config matrix, there should be a more aggressive matrix compared to the first matrix versions)

I am planning to migrate to open-meteo as weather-api as there is no API key needed and individual Panel Azimith / Panel Tilt can be part of the data. This should be more accurate.

https://open-meteo.com/en/docs#minutely_15=sunshine_duration,global_tilted_irradiance&hourly=temperature_2m,snow_depth,global_tilted_irradiance&daily=sunrise,sunset,daylight_duration,sunshine_duration,snowfall_sum,shortwave_radiation_sum&timezone=Europe%2FBerlin&forecast_days=1&forecast_minutely_15=96&tilt=35&azimuth=40

Regards
Christian

bacanol, 2024-01-30:
@christian1980nrw
Thank you for your message. I think, we misunderstood each other at least ab it. 

It is not an issue of discharging the battery in general. The issue is that the battery get charged via sun and either AC-inverters or DC-charge-controllers or both over the outline of the day. If the battery is nearly full and the ESS setting 7 is active with 0 discharge allowed, it reduces the power of the DC-charge-controllers as they are only allowed to charge the battery. Any energy that is produced that exceeds the load capacity of the battery is lost. 

I guess the issue is more related to the ESS setting 7 with a discharge allowance of 0 which is preventing the sun-power from DC-charge-crontrollers to be consumed else than in the battery charge. Maybe it is an issue with the Victron environment as such, but it keeps to be an issue. Maybe there is a way to only use the ESS 7 setting if the battery is 80% charged?

Regards,
Nico

christian1980nrw, 2024-01-30:
@bacanol 
Ah ok, i only have a AC-AC system, so i dont have that effect.
So there is an additional check needed to re-activate the inverter if the battery is getting full...
OK, i will implement this within the next days. My problem is only lack of time.

Regards Christian

christian1980nrw, 2024-01-31:
@bacanol 
Teste mal bitte die Änderung. Habe die Schwelle auf 90% gesetzt und vorerst hart im Code in Zeile 1499.
Ich glaube morgen wird es wieder sonnig.
https://raw.githubusercontent.com/christian1980nrw/Spotmarket-Switcher/dev/scripts/controller.sh
Falls positives Feedback von Dir kommt, übernehme ich den Parameter in eine neue Config Version.

Gruß!
Christian

christian1980nrw, 2024-02-02:
@bacanol Hattest Du schon Zeit zum testen?

bacanol, 2024-02-03:
Moin @christian1980nrw ,

Leider noch nicht. Die Bedingungen waren hier noch nicht wieder gegeben. Ich hoffe, dass es in der neuen Woche passen könnte. 

Ich melde mich. 
Viele Grüße 😉 Nico

christian1980nrw, 2024-02-03:
Ich habe die Abbruchfunktionen gestern nochmal umgebaut und den Code an der Stelle einfacher lesbar gemacht.
War vorher etwas zu kompliziert verschachtelt. Der Teil stammte aus einem Pull-Request eines Forks.

bacanol, 2024-03-10:
Hallo @christian1980nrw,

bitte entschuldige die späte Rückmeldung. Mit der DEV-Version funktioniert es bei mir. Der Effekt, bei strahlendem Sonnenschein keinen DC Strom mehr zu produzieren oder zu verkaufen ist nicht mehr eingetreten.

Ich laufe jetzt erstmal mit der DEV weiter bis Du ein neues Release machst. ;-)

Danke und sonnige Grüße
Nico

closed 2024-04-05 by christian1980nrw