butterwecksolutions/Spotmarket-Switcher #184: Error with Tibber pricing API?

kLOsk, 2024-09-13:
There appears to be an error with the tibber api pricing interpretation. I'm seeing the following logs

```
Sep 13 04:00:01 einstein cron.info CROND[28563]: (root) CMD (/data/etc/Spotmarket-Switcher/controller.sh)
Sep 13 04:00:02 einstein cron.info CROND[28562]: (root) CMDOUT (^[[1;32mI: Bash Version: GNU bash, version 5.0.18(1)-release (arm-ve-linux-gnueabi)^[[0m)
Sep 13 04:00:02 einstein cron.info CROND[28562]: (root) CMDOUT (^[[1;32mI: Spotmarket-Switcher - Version 2.4.10^[[0m)
Sep 13 04:00:09 einstein cron.info CROND[28562]: (root) CMDOUT ( [|]  Loading... [/]  Loading... [-]  Loading... [\]  Loading... [|]  Loading... [/]  Loading... [-]  Loading... [\]  Loading... [|]  Loading... [/]  Loading... [-]  Loading... [\]  Loading... [|]  Loading.
Sep 13 04:00:12 einstein cron.info CROND[28562]: (root) CMDOUT (. [|]  Loading... [/]  Loading... [-]  Loading... [\]  Loading... [|]  Loading... [/]  Loading... [-]  Loading... [\]  Loading... [|]  Loading... [/]  Loading... [-]  Loading... [\]  Loading... [|]  Loading
```

I have now switched to awattar and everything seems to be working.

Not sure how to debug the Tibber error as no error is displayed anywhere:

```
./controller.sh 
I: Bash Version: GNU bash, version 5.0.18(1)-release (arm-ve-linux-gnueabi)
I: Spotmarket-Switcher - Version 2.4.10
 [|]  Loading...Config validation passed.
```

christian1980nrw, 2024-09-13:
Hi, i wil check it with my Tibber Api Key within the next days. Maybe they changed something.

christian1980nrw, 2024-09-15:
Cannot confirm the issue. Tibber API is still working perfect.

closed 2024-09-15 by christian1980nrw